NordicSemiconductor/pc-nrfutil #1: behave package missing from requirements.txt

hakonfam, 2015-08-28:
When installing version 0.3 on a mac I got an error message when running "python setup.py install" saying that "setuptools_behave is missing".

This was fixed by adding a dependency to the "behave" package in requirements.txt.

follower, 2015-09-10:
While the "fix" suggested has been applied with baaf254185516dd3d0d2b8932f145e63f246a66e I think a better option would be to remove the import line that causes the error https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/setup.py#L44 because the only use is commented out: https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/setup.py#L112.

As `tests_require` already specifies `behave` is required for testing (and only testing AFAICT) it shouldn't need to be placed in `requirements.txt`. (Although I admit I'm not sure how the `bdd_test` can be added in future in that scenario.)

Also, it seems some or all of the current content in `requirements.txt` could/should be included in `install_requires` instead: https://packaging.python.org/en/latest/requirements.html This should enable `pip` to handle the dependencies automatically rather than have a two-step installation process.

There might also be some benefit from declaring certain dependencies (e.g. `pyserial`) as only required for certain features: http://pythonhosted.org//setuptools/setuptools.html#declaring-extras-optional-features-with-their-own-dependencies

ndahlquist, 2016-06-07:
This is resolved by https://github.com/NordicSemiconductor/pc-nrfutil/commit/4f2c873c89e8128012d2ed9018467342fffeeca2; we can probably close this issue. Thanks!

closed 2016-06-10 by carlescufi