NordicSemiconductor/pc-nrfutil #4: Fails on Mac OS Capitan

janekm, 2015-12-22:
I'm getting the following error message:

nrfutil dfu genpkg respeck.zip --application _build/nrf51822_xxaa_s110.hex --application-version 0xffff --dev-revision 0xffff --dev-type 0xffff --sd-req 0xfffe
Traceback (most recent call last):
  File "/Users/janek/Library/Python/2.7/bin/nrfutil", line 9, in <module>
    load_entry_point('nrfutil==0.3.0', 'console_scripts', 'nrfutil')()
  File "/Library/Python/2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/janek/Library/Python/2.7/lib/python/site-packages/nrfutil-0.3.0-py2.7.egg/nordicsemi/__main__.py", line 236, in genpkg
    package.generate_package(zipfile_path)
  File "/Users/janek/Library/Python/2.7/lib/python/site-packages/nrfutil-0.3.0-py2.7.egg/nordicsemi/dfu/package.py", line 220, in generate_package
    init_packet = self._create_init_packet(firmware)
  File "/Users/janek/Library/Python/2.7/lib/python/site-packages/nrfutil-0.3.0-py2.7.egg/nordicsemi/dfu/package.py", line 320, in _create_init_packet
    return p.generate_packet()
  File "/Users/janek/Library/Python/2.7/lib/python/site-packages/nrfutil-0.3.0-py2.7.egg/nordicsemi/dfu/init_packet.py", line 79, in generate_packet
    format_string = self.__generate_struct_format_string()
  File "/Users/janek/Library/Python/2.7/lib/python/site-packages/nrfutil-0.3.0-py2.7.egg/nordicsemi/dfu/init_packet.py", line 97, in __generate_struct_format_string
    for key in sorted(self.init_packet_fields.keys(), key=lambda x: x.value):
  File "/Users/janek/Library/Python/2.7/lib/python/site-packages/nrfutil-0.3.0-py2.7.egg/nordicsemi/dfu/init_packet.py", line 97, in <lambda>
    for key in sorted(self.init_packet_fields.keys(), key=lambda x: x.value):
AttributeError: 'int' object has no attribute 'value'

kenr, 2015-12-22:
I'm not able to reproduce this error. Are you able to share the .hex file ?

janekm, 2015-12-23:
Sure, here it is:

[nrf51822_xxaa_s110.hex.zip](https://github.com/NordicSemiconductor/pc-nrfutil/files/70668/nrf51822_xxaa_s110.hex.zip)

aep, 2016-03-16:
same error here on linux

aep, 2016-03-16:
this fixes it for me:

    --- ./build/lib/nordicsemi/dfu/init_packet.py	2016-03-16 14:43:30.000000000 +0100
    +++ /usr/lib/python2.7/site-packages/nrfutil-0.3.0-   py2.7.egg/nordicsemi/dfu/init_packet.py	2016-03-16 14:55:42.765861290 +0100
    @@ -79,7 +79,7 @@
             format_string = self.__generate_struct_format_string()
             args = []
  
    -        for key in sorted(self.init_packet_fields.keys(), key=lambda x: x.value):
    +        for key in sorted(self.init_packet_fields.keys(), key=lambda x: x):
                # Add length to fields that required that
                 if key in [PacketField.REQUIRED_SOFTDEVICES_ARRAY,
                            PacketField.OPT_DATA]:
    @@ -94,7 +94,8 @@
             format_string = "<"  # Use little endian format with standard sizes for python,
             # see https://docs.python.org/2/library/struct.html
     
    -        for key in sorted(self.init_packet_fields.keys(), key=lambda x: x.value):
    +        for key in sorted(self.init_packet_fields.keys(), key=lambda x: x):
                 if key in [PacketField.DEVICE_TYPE,
                            PacketField.DEVICE_REVISION,
                            ]:

closed 2016-06-23 by carlescufi