NordicSemiconductor/pc-nrfutil #9: Attribute Error when running compiled .exe

almithani, 2016-04-25:
Hello,

I'm able to successfully run nrfutil using python, but I'm having a little trouble when running my own compiled nrfutil.exe.  Here's my traceback:

V:\Code\Firmware\pc-nrfutil-master\0.3.0>nrfutil.exe
Traceback (most recent call last):
  File "__main__.py", line 32, in <module>
  File "zipextimporter.pyc", line 82, in load_module
  File "click\__init__.pyc", line 18, in <module>
  File "zipextimporter.pyc", line 82, in load_module
  File "click\core.pyc", line 7, in <module>
  File "zipextimporter.pyc", line 82, in load_module
  File "click\types.pyc", line 4, in <module>
  File "zipextimporter.pyc", line 82, in load_module
  File "click\_compat.pyc", line 530, in <module>
  File "zipextimporter.pyc", line 82, in load_module
  File "click\_winconsole.pyc", line 23, in <module>
  File "ctypes\__init__.pyc", line 378, in __getattr__
  File "ctypes\__init__.pyc", line 383, in __getitem__
AttributeError: function 'PyObject_GetBuffer' not found

carlescufi, 2016-06-29:
is this still an issue on 0.5.1 or 1.2.5?

carlescufi, 2016-07-05:
Closing due to inactivity. Reopen if still reproducible.

ACGarland, 2016-08-12:
FWIW, I'm seeing this same problem using 0.5.1.

(I'm using 0.5.1 because we are using serial dfu.)

ACGarland, 2016-08-12:
I would reopen this issue, but lack the permission to do so.  I'm not sure I should open a new issue and reference this one instead?  In any case, my trace output is identical to what is shown above.

ACGarland, 2016-08-12:
Here's something I did on a hunch to work around this problem...

The problem appears to be related to how click refers to the GetBuffer function, perhaps they fail to account for C++ [name mangling](http://stackoverflow.com/a/8473993/3174688)?  So I took a chance and thought I'd try regressing to previous versions of click.

pip uninstall click
pip install click==5.0
python setup.py py2exe

Then, it failed for lack of enum package.

pip install enum
python setup.py py2exe

Voila! It now works properly without any error.

Hope this tidbit might prove useful.

carlescufi, 2016-08-13:
I reopened now but I see you found a workaround. I will leave it open until we get a better solution for 0.5.1. Once users start switching to the master version this shouldn't be a problem anymore since they will be expected to install from pip, which automagically creates an .exe.

ACGarland, 2016-08-15:
Thanks Carles.

I wonder if installation from pip will solve the problem for folks who have customized nrfutil--won't we still need to make it through the python-to-EXE conversion process using our customizations?

In my case, I added a "-r" or "--record" flag to capture the HCI/SLIP packet output into a binary file (it uses the package file path and appends an '.hci' extension for the binary output file). That allows me to "replay" that file through a simpler HCI/SLIP implementation on an embedded processor connected to the final serial-enabled bootloader.  That way, we can still do serial DFU from a very limited environment and push the complexity of all that nrfutil has to do "upstream" to be on a PC or Linux box where there is good support for python and the package dependencies.

At any rate, as you've observed, I'm able to proceed with the workaround.  My customized version is now called at the end of the build from Keil MDK to generate the requisite binary recording which gets packaged up as part of the build process.

carlescufi, 2016-08-16:
Yes, this will indeed still be a problem for customized versions of nrfutil as you mention. Thanks for all your observations and tips, we will try to figure this out.

ACGarland, 2016-08-17:
Hi Carles,

Not sure whether this is something that should be opened as an issue/bug?  I've found, when testing with my experimental version which supports recording the packet output stream without a serial bootloader connected/present, that there may be a defect related to packet sequence numbering for the case where the DFU ZIP file used for the DFU update contains more than one image.

For example, if nrfutil is first used to build a DFU ZIP file containing bootloader, softdevice, and application, then the resulting file contains two images: 1) a BL+SD image, and 2) an APP image.

If this same DFU ZIP file is provided to nrfutil during serial DFU,
I noticed in my testing that the serial transport does not reset the packet sequence number between sending the two images.  This results in a somewhat "random" staring sequence number for the second image (depending on how many packets are in the first image modulus 8). In my case, I was producing two HCI recording files, but the second one never was accepted correctly since it didn't begin with sequence 1.  I corrected the problem by setting the HCiPacket.sequence_number = 0 within the serial transport open() function.  (Maybe close would have been a better location, I'm not sure.)  In any case, that produced two recording files where both of them have sequence 1 for the first packet and that seems to work.

Thought I'd pass that along...

Also: in my variation, when I'm recording only (no COM port specified), then it skips the 1 second and 10 second delays it normally inserts when talking to a real bootloader. This makes our build process more efficient since we generate our recording/HCI files as part of the build process and so don't incur those delays each time.

carlescufi, 2016-08-17:
@ACGarland yes, can you report this as a separate issue please? I will then remove your comment from this thread, I want to fix py2exe first

carlescufi, 2016-08-17:
https://github.com/pallets/click/issues/635

carlescufi, 2016-08-17:
I've tagged 0.5.2 to solve this issue. I am going to close this for now, @ACGarland please open a new one for the HCI behaviour

closed 2016-08-17 by carlescufi

ACGarland, 2016-08-17:
[pc-nrfutil/#29](https://github.com/NordicSemiconductor/pc-nrfutil/issues/29)

carlescufi, 2016-08-17:
The modern DFU (master) now uses PyInstaller instead of py2exe to avoid these kind of issues. For the legacy version, 0.5.2 should now work fine with py2exe.