NordicSemiconductor/pc-nrfutil #13: Q: SDK versions?

jbech-linaro, 2016-06-09:
Hi,

I'm a bit confused regarding the SDK versions. In the [versions](https://github.com/NordicSemiconductor/pc-nrfutil#versions) section, you mention that there are two ways of generating firmware packages. One for SDK <= v11.0  and one for SDK >= v11.1.

But when I look for the SDK at [download pages](http://www.nordicsemi.com/eng/Products/Bluetooth-low-energy2/nRF5-SDK), I can only find the v11.0? Would you mind explaining what am I missing? Likewise I can only find documentation up to v11.0 at the [documentation pages](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk%2Fdita%2Fsdk%2Fnrf5_sdk.html&cp=5_0).

bihanssen, 2016-06-09:
Hi,  
SDK v11.1 has not been released yet. The firmware package description is just a bit ahead to be ready for the next SDK release.

jbech-linaro, 2016-06-09:
Hei Bjørn,

I see, thanks for clarifying this. I'm closing the issue.

// Joakim

closed 2016-06-09 by jbech-linaro