NordicSemiconductor/pc-nrfutil #21: Auto flashing does not work when performing DFU over BLE

hakonfam, 2016-06-24:
Using version 5a5e418f49579bd with pc-ble-driver-py version 0.4.0, the following happens when attempting to perform a DFU over BLE with auto-flashing of the connectivity driver:

```
mypath>nrfutil dfu ble -f -p COM40 -pkg app_v1.0.0.zip
No target selected. Default device name: DfuTarg is used.
  [------------------------------------]    0%

Failed to update the target. Error is:
```

carlescufi, 2016-06-24:
The problem is that we cannot know if it was a flashing error or a transfer error. I vote to move out the flashing functionality from DfuTransportBle into __main__ so that we can be more precise in reporting

WojciechJasko, 2016-06-24:
Can you try [aa0372b](https://github.com/NordicSemiconductor/pc-nrfutil/commit/aa0372b048ebe223577f8748bd5bb905e6fe8373)

carlescufi, 2016-06-29:
This is now fixed

closed 2016-06-29 by carlescufi