NordicSemiconductor/pc-nrfutil #25: Unable to perform DFU upgrade over serial

jsiverskog, 2016-07-05:
Hi!
I have an nRF52 evaluation board which I want to test DFU upgrade over serial on.

Since I'm using SDK 11, I'm checking out branch 0_5_1.

In README.md it is stated "Note: DFU over a serial line is currently disabled". Is this true? Then what exactly is this project for? The nrfutil command line tool doesn't say anything that it shouldn't work.

I'm flashing SoftDevice s132 2.0.1 and I've tried both examples/dfu/bootloader/pca10040/single_bank_serial_s132/armgcc and examples/dfu/bootloader/pca10040/dual_bank_serial_s132/armgcc, but I can't get the tool to work. I get this:
```
nrfutil dfu serial --package ../dfu_test_app_hrm_s132.zip -p /dev/ttyACM0  --flowcontrol --baudrate 38400
Upgrading target on /tmp/ser with DFU package /root/dfu_test_app_hrm_s132.zip. Flow control is enabled.
  [------------------------------------]    0%Timed out waiting for acknowledgement from device.

Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.
```

When sniffing on the serial communication using socat, I get:
```
[root@XXX ~]# socat -d -v -x PTY,link=/tmp/ser,wait-slave,raw /dev/ttyACM0,raw
> 2016/07/05 15:00:27.776812  length=28 from=0 to=27
 c0 d1 4e 01 e0 03 00 00 00 04 00 00 00 00 00 00  ..N.............
 00 00 00 00 00 e0 4b 00 00 dd c6 c0              ......K.....
--
```

Thanks, Jacob

carlescufi, 2016-07-05:
serial DFU is disabled on master only. For legacy DFU (i.e. 0.5.1, serial DFU is functional). See the README.md corresponding to 0.5.1 for further details.

Are you using SDK 11.0?

jsiverskog, 2016-07-06:
Yes I'm using SDK 11.

Are the Softdevice and bootloaders I'm programming right?

Can I get more debug information somehow?

vidarbe, 2016-07-06:
Hi Jacob,

Have you tried running as root? It may be required to access the COM port. 

Here's the command I used (didn't build it so I just ran the script directly): 
sudo python __main__.py dfu serial -pkg application.zip -p /dev/ttyACM0 -fc

carlescufi, 2016-07-06:
For more info on serial ports and permissions:
https://github.com/NordicSemiconductor/pc-ble-driver#j-link-usb-cdc-serial-ports

carlescufi, 2016-07-07:
Please let us know how you get on with this.

crespum, 2016-07-19:
I'm having the very same problem with SDK 10. I tweaked nrfutils 0.5.1 to show the received and the transmitted packets and I noticed that it was sending packets bigger than 512 bytes after encoding. I tried reducing the maximum size but it didn't solve the issue.

```
Upgrading target on /dev/ttyACM0 with DFU package dfu_test_app_hrm_s130.zip. Flow control is enabled.
  [------------------------------------]    0%Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 20524
Sending DFU start packet, afterwards we wait for the flash on target to be initialized before continuing.
PC -> target: c0d14e01e0030000000400000000000000000000002c500000d989c0
PC <- target: c0100000f0c0
Sending DFU init packet
PC -> target: c0da4e01d701000000ffffffffffffffff01006700e6970000e517c0
PC <- target: c0180000e8c0
Sending firmware file
  [------------------------------------]    1%  00:18:16PC -> target: c0e34e1eb104000000dbdc39002049c1010063c1010065c101000000000000000000000000000000000000000000000000000000000067c10100000000000000000069c101006bc101006dc101006dc10100ddc401006dc101006dc101000000000099c201006dc101006dc101006dc101006dc101006dc101006dc101006dc101006dc101006dc101006dc10100f1c301006dc101006dc1010011c401006dc1010065c401006dc101006dc101006dc101000000000000000000000000000000000000000000000000000348854600f0b8f80048004709fa0100dbdc390020401e00bf00bf00bf00bf00bf00bf00bf00bf00bf00bf00bf00bff1d170470000401e00bf00bf00bf00bf00bf00bf00bf00bf00bf00bf00bf00bff1d170470000056885f308884068ff2464b2eff30585002d01d1a64600472546064621273fbaf0b40024002500260027f0b4f92040b2004700000321094802680a430260084802680a4302600748804707480047fee7fee7fee7fee7fee7fee70000240500405405004089c40100c1dbdc010030b50b46014600202022012409e00d46d5409d4205d31d469540491b2546954040191546521e002df1dc30bd10b5431a934209d28318881803e0401e01785b1e1970521ef9d210bd03460b439b0703d009e008c9121f08dbdc042afad203e00b78d766c0
Timed out waiting for acknowledgement from device.

Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.

Possible causes:
- bootloader, SoftDevice or application on target does not match the requirements in the DFU package.
- baud rate or flow control is not the same as in the target bootloader.
- target is not in DFU mode. If using the SDK examples, press Button 4 and RESET and release both to enter DFU mode.
```

If I try building and programming the bootloader, softdevices and the example provided with SDK 11, I get:
```
Upgrading target on /dev/ttyACM0 with DFU package /home/xabi/Documents/Dev/nRF51/build/dfu_test_app_hrm_s130.zip. Flow control is enabled.
  [------------------------------------]    0%Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 18904
Sending DFU start packet, afterwards we wait for the flash on target to be initialized before continuing.
PC -> target: c0d14e01e003000000040000000000000000000000d84900009701c0
PC <- target: c0100000f0c0
Sending DFU init packet
PC -> target: c0da4e01d701000000ffffffffffffffff01007800b9a000008229c0
Timed out waiting for acknowledgement from device.
PC <- target: c0ff

Failed to upgrade target. Error is: list index out of range

Possible causes:
- bootloader, SoftDevice or application on target does not match the requirements in the DFU package.
- baud rate or flow control is not the same as in the target bootloader.
- target is not in DFU mode. If using the SDK examples, press Button 4 and RESET and release both to enter DFU mode.
```

coolnumber9, 2016-08-16:
Hi All,

I'm having the same issue as Jacob @jsiverskog and Xabi @crespum using **nrfutil v0.3.0.0** and **nRF52 DK (PCA10040)**.

My initial observation is that it won't work on the pre-built zip files included in **SDK 11**. 

Here's what I did to run the DFU serial example:

1. Run **nRFgo Studio v1.21.2.10**
2. On the **Device Manager**, selected **nRFx Bootloader**
3. On the **DFU using serial bootloader**, browsed the sample HEX file (_not the zip package_). I selected the **dfu_test_app_hrm_s132.hex** from the pre-built hex samples directory: (**_nRF5_SDK_11.0.0_89a8197\examples\dfu\hci_dfu_send_hex\test_images_dual_bank_update_nrf52_**) 
4. Selected the appropriate **COM port**.
5. Selected baudrate (I used **38400**).
6. Checked the **flowcontrol** checkbox.
7. Clicked **Program**.
8. After successfully programming nRF52, I checked the log window on nRFgo Studio and browsed to the directory where nRFgo Studio generated the **zip file package** from the hex file being selected in Step 3.
9. In the log window, I observed that nRFgo Studio called **nrfutil.exe dfu genpkg** and dumps in the the Temp directory. In my Windows system, it is stored in **../AppData/Local/Temp/** with the file name of **TempDFUPackage_hN6900.zip**.
10. Pressed Button 4 of nRF52 DK and then pressed RESET button.
11. LED 3 is turned ON which means it's on the bootloader mode already.
12. Then in the terminal, I ran the nrfutil (_not nRFgo Studio this time_) with the following command and it worked fine: **nrfutil.exe dfu serial --package=C:/Users/coolnumber9/AppData/Local/Temp/TempDFUPackage_hN6900-test.zip --port=COM4 --baudrate=38400 --flowcontrol**.

I haven't validated if the app firmware running is the new one. I will continue to investigate. Let me know it goes to your system as well.

Thanks,
Kristoffer

crespum, 2016-08-17:
@coolnumber9 my problem was related to the OS. I've followed the same steps on Windows and it worked. I have to check the serial port configuration, but right now I don't want to spend time on this.

FYI, some more discussion is going on [here](https://devzone.nordicsemi.com/question/88365/problem-with-uart-dfu/).

jsiverskog, 2016-08-17:
Hi!
Sorry for my late reply - I've been away on vacation.

I'm pretty sure I tried as root as well, so it shouldn't be any serial port permission issue (@vidarbe, @carlescufi).

I can't even get it to work on Windows with nRFgo Studio when following the instructions from @coolnumber9 (same error message as on Linux).

Nordic people, have you managed to perform a DFU upgrade over serial lately?

Thanks, Jacob

bihanssen, 2017-03-22:
@jsiverskog As serial DFU was added to nrfutil master (non-legacy) a while back, are you able to use master instead?

jsiverskog, 2017-03-22:
@bihanssen I have written a custom serial DFU transport layer, so this issue is not relevant to me anymore.

closed 2017-03-22 by jsiverskog