NordicSemiconductor/pc-nrfutil #27: Issue running on OS X on El-Capitan with python2.7

vrahane, 2016-07-30:
{noformat}
Vipuls-MacBook-Pro:pc-nrfutil vipul$ nrfutil keys generate private.pem
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6

Vipuls-MacBook-Pro:pc-nrfutil vipul$ python
Python 2.7.11 (default, Jan 22 2016, 08:29:18) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.2 (clang-700.1.81)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> 
{noformat}

carlescufi, 2016-07-31:
Did you install with pip?

carlescufi, 2016-07-31:
Also, are you running the built-in OS X Python or a Homebrew one? Try:

`$ which python`

vrahane, 2016-07-31:
Yes, I did.
`Vipuls-MacBook-Pro:myproj vipul$ pip -V
pip 8.1.2 from /usr/local/lib/python2.7/site-packages (python 2.7)`

carlescufi, 2016-07-31:
This could be a library problem. Can you give me the ouput of:

`otool -L /Library/Python/2.7/site-packages/pc_ble_driver_py/lib/macos_osx/_pc_ble_driver.so`
and
`otool -L /Library/Python/2.7/site-packages/pc_ble_driver_py/lib/macos_osx/libpc_ble_driver_shared.dylib`

?

vrahane, 2016-07-31:
```
Vipuls-MacBook-Pro:myproj vipul$ otool -L /Library/Python/2.7/site-packages/pc_ble_driver_py/lib/macos_osx/_pc_ble_driver.so
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/otool: can't open file: /Library/Python/2.7/site-packages/pc_ble_driver_py/lib/macos_osx/_pc_ble_driver.so (No such file or directory)
```

and

```
Vipuls-MacBook-Pro:myproj vipul$ otool -L /Library/Python/2.7/site-packages/pc_ble_driver_py/lib/macos_osx/libpc_ble_driver_shared.dylib
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/otool: can't open file: /Library/Python/2.7/site-packages/pc_ble_driver_py/lib/macos_osx/libpc_ble_driver_shared.dylib (No such file or directory)
```

carlescufi, 2016-07-31:
Oh right I see, your python libs are in /usr/local/lib so you're using Homebrew's Python. Not sure why, but apparently you need to use the default OS X python build for our libs to work. 

You can simply edit `/etc/paths` to put `/usr/bin` before `/usr/local/bin` then install again with pip (but using the default python and not homebrew's).

carlescufi, 2016-07-31:
More info about switching between Hombrew's Python and the default OS X Python can be found [here](http://stackoverflow.com/questions/5157678/python-homebrew-by-default)

vrahane, 2016-07-31:
Sure, I will try it out and let you know. Thank you.

vrahane, 2016-07-31:
So, I had to do a bunch of things but it worked finally. I had to reinstall pip and nrfutil after removing brew installed python. Thank you so much for your help.

closed 2016-07-31 by vrahane

carlescufi, 2016-07-31:
No prob, thanks for reporting it. I'll see if I can add this to the doc.

vikeri, 2017-05-11:
Getting the same issue, using the built in macOS Python is not an alternative for me unfortunately. This only occurs when I run `nrfutil dfu ble` it works as expected when I run `nrfutil pkg generate`.