NordicSemiconductor/pc-nrfutil #28: ImportError: cannot import name enum_type_wrapper

eliotstock, 2016-08-04:
I'm on Mac OSX 10.11, using Python installed from www.python.org, and using master of this project.

```
$ python --version
Python 2.7.12

$ pip --version
pip 8.1.2 from /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (python 2.7)

$ sudo -H python setup.py install
[installs with no errors]
```

Because of this issue in Google's protobuf https://github.com/google/protobuf/issues/737, I modify the perms on one of protobuf's directories with `sudo chmod 644 /Library/Python/2.7/site-packages/protobuf-3.0.0-py2.7.egg/EGG-INFO/*`

```
$ nrfutil --help
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/bin/nrfutil", line 9, in <module>
    load_entry_point('nrfutil==1.4.5', 'console_scripts', 'nrfutil')()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 357, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 2394, in load_entry_point
    return ep.load()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 2108, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/nrfutil-1.4.5-py2.7.egg/nordicsemi/__main__.py", line 47, in <module>
    from nordicsemi.dfu.dfu import Dfu
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/nrfutil-1.4.5-py2.7.egg/nordicsemi/dfu/dfu.py", line 47, in <module>
    from nordicsemi.dfu.package         import Package
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/nrfutil-1.4.5-py2.7.egg/nordicsemi/dfu/package.py", line 52, in <module>
    from nordicsemi.dfu.init_packet_pb import *
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/nrfutil-1.4.5-py2.7.egg/nordicsemi/dfu/init_packet_pb.py", line 37, in <module>
    import dfu_cc_pb2 as pb
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/nrfutil-1.4.5-py2.7.egg/nordicsemi/dfu/dfu_cc_pb2.py", line 42, in <module>
    from google.protobuf.internal import enum_type_wrapper
ImportError: cannot import name enum_type_wrapper
```

Is this something to do with the version of protobuf you're using?

carlescufi, 2016-08-04:
Have you installed with pip? if not, have you installed all dependencies? They are listed in here:

https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/setup.py#L84

Also, we haven't tested on OS X with a Python installation different from the default one (the one built-in with the OS)

eliotstock, 2016-08-04:
I've installed from source, because I want the latest, not the one in PyPI. But I have all the dependencies, in versions on or later than the ones you require:

```
$ sudo -H pip install pyserial
Requirement already satisfied (use --upgrade to upgrade): pyserial in /Library/Python/2.7/site-packages/pyserial-3.1.1-py2.7.egg
eliot-macbook-pro:dev Eliot$ sudo -H pip install enum34
Requirement already satisfied (use --upgrade to upgrade): enum34 in /Library/Python/2.7/site-packages/enum34-1.1.6-py2.7.egg
eliot-macbook-pro:dev Eliot$ sudo -H pip install click
Requirement already satisfied (use --upgrade to upgrade): click in /Library/Python/2.7/site-packages/click-6.6-py2.7.egg
eliot-macbook-pro:dev Eliot$ sudo -H pip install click
Requirement already satisfied (use --upgrade to upgrade): click in /Library/Python/2.7/site-packages/click-6.6-py2.7.egg
eliot-macbook-pro:dev Eliot$ sudo -H pip install ecdsa
Requirement already satisfied (use --upgrade to upgrade): ecdsa in /Library/Python/2.7/site-packages/ecdsa-0.13-py2.7.egg
eliot-macbook-pro:dev Eliot$ sudo -H pip install behave
Requirement already satisfied (use --upgrade to upgrade): behave in /Library/Python/2.7/site-packages/behave-1.2.5-py2.7.egg
Requirement already satisfied (use --upgrade to upgrade): parse>=1.6.3 in /Library/Python/2.7/site-packages/parse-1.6.6-py2.7.egg (from behave)
Requirement already satisfied (use --upgrade to upgrade): parse_type>=0.3.4 in /Library/Python/2.7/site-packages/parse_type-0.3.4-py2.7.egg (from behave)
Requirement already satisfied (use --upgrade to upgrade): six in /Library/Python/2.7/site-packages/six-1.10.0-py2.7.egg (from behave)
Requirement already satisfied (use --upgrade to upgrade): enum34 in /Library/Python/2.7/site-packages/enum34-1.1.6-py2.7.egg (from parse_type>=0.3.4->behave)
eliot-macbook-pro:dev Eliot$ sudo -H pip install protobuf
Requirement already satisfied (use --upgrade to upgrade): protobuf in /Library/Python/2.7/site-packages/protobuf-3.0.0-py2.7.egg
Requirement already satisfied (use --upgrade to upgrade): six>=1.9 in /Library/Python/2.7/site-packages/six-1.10.0-py2.7.egg (from protobuf)
Requirement already satisfied (use --upgrade to upgrade): setuptools in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from protobuf)
eliot-macbook-pro:dev Eliot$ sudo -H pip install pc_ble_driver_py
Requirement already satisfied (use --upgrade to upgrade): pc_ble_driver_py in /Library/Python/2.7/site-packages/pc_ble_driver_py-0.5.0-py2.7.egg
Requirement already satisfied (use --upgrade to upgrade): enum34 in /Library/Python/2.7/site-packages/enum34-1.1.6-py2.7.egg (from pc_ble_driver_py)
Requirement already satisfied (use --upgrade to upgrade): wrapt in /Library/Python/2.7/site-packages/wrapt-1.10.8-py2.7-macosx-10.11-intel.egg (from pc_ble_driver_py)
Requirement already satisfied (use --upgrade to upgrade): future in /Library/Python/2.7/site-packages/future-0.15.2-py2.7.egg (from pc_ble_driver_py)
```

I'd be happy to use the OS version of Python, but then how do I install pip? The installation instructions for pip at https://pip.pypa.io/en/stable/installing/ are pretty clear I should be using a www.python.org version of Python:

> Warning
> 
> Be cautious if you're using a Python install that's managed by your operating system or another package manager. get-pip.py does not coordinate with those tools, and may leave your system in an inconsistent state.

Happy to replicate the pip install on your machines if you can provide instructions though.

carlescufi, 2016-08-04:
first things first, do you want the latest nrfutil because you're using a preview of SDK 12? Because if you're using SDK 11 you should be using the tagged 0.5.1 version

eliotstock, 2016-08-04:
Yep, aware of that. Not on an SDK 12 preview yet (although I'd like to be). Just trying to get prepared for it and will then roll back to 0.5.1 and get the rest of it working (bootloader, app support for DFU, etc.) on SDK 11. I can wait, but I thought you'd like to know about any issues now.

carlescufi, 2016-08-04:
OK, thanks for the issue report! Let's start by using pip on the default OS X python install. Never heard of anyone having problems with that, so let's give it a go.
First make sure that `which python` returns `/usr/bin/python`. For more info check here: 
https://github.com/NordicSemiconductor/pc-nrfutil/issues/27
Then install pip:
http://stackoverflow.com/a/29188667/337201

eliotstock, 2016-08-04:
OK, I've successfully removed the www.python.org version of Python and easy_install appears to install OK, but then easy_install is unable to install pip:

```
$ sudo easy_install pip
Traceback (most recent call last):
  File "/usr/local/bin/easy_install", line 9, in <module>
    load_entry_point('setuptools==25.1.4', 'console_scripts', 'easy_install')()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 357, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 2394, in load_entry_point
    return ep.load()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 2108, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "build/bdist.macosx-10.11-intel/egg/setuptools/__init__.py", line 11, in <module>
  File "build/bdist.macosx-10.11-intel/egg/setuptools/extern/__init__.py", line 1, in <module>
ImportError: No module named extern
```

It seems that easy_install is not installed properly:

```
$ sudo easy_install
Traceback (most recent call last):
  File "/usr/local/bin/easy_install", line 9, in <module>
    load_entry_point('setuptools==25.1.4', 'console_scripts', 'easy_install')()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 357, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 2394, in load_entry_point
    return ep.load()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 2108, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "build/bdist.macosx-10.11-intel/egg/setuptools/__init__.py", line 11, in <module>
  File "build/bdist.macosx-10.11-intel/egg/setuptools/extern/__init__.py", line 1, in <module>
ImportError: No module named extern
```

Perhaps it's related to this issue in pip: https://github.com/pypa/setuptools/issues/514

carlescufi, 2016-08-04:
Right. Can you try downloading: https://bootstrap.pypa.io/get-pip.py and running `python get-pip.py` ?

eliotstock, 2016-08-04:
Sure. That failed when run as me (`OSError: [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/pip'`) but worked when sudo'ed (with some warnings about directory ownership).

I can now run `pip list` OK, without sudo.

carlescufi, 2016-08-04:
Great, were you able to run nrfutil with the stock vanilla Python?

eliotstock, 2016-08-04:
No. It installs OK, but doesn't run. Same error as above.

In your README.md you mention the need to update setuptools to the latest version. This fails for me. Not sure if that's the cause.

```
sudo pip install -U setuptools
The directory '/Users/Eliot/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/Eliot/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting setuptools
  Downloading setuptools-25.1.4-py2.py3-none-any.whl (443kB)
    100% |████████████████████████████████| 450kB 565kB/s 
Installing collected packages: setuptools
  Found existing installation: setuptools 1.1.6
    Uninstalling setuptools-1.1.6:
Exception:
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Library/Python/2.7/site-packages/pip/commands/install.py", line 317, in run
    prefix=options.prefix_path,
  File "/Library/Python/2.7/site-packages/pip/req/req_set.py", line 736, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Python/2.7/site-packages/pip/req/req_install.py", line 742, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Python/2.7/site-packages/pip/req/req_uninstall.py", line 115, in remove
    renames(path, new_path)
  File "/Library/Python/2.7/site-packages/pip/utils/__init__.py", line 267, in renames
    shutil.move(old, new)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 299, in move
    copytree(src, real_dst, symlinks=True)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 208, in copytree
    raise Error, errors
Error: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', "[Errno 1] Operation not permitted: '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', "[Errno 1] Operation not permitted: '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', "[Errno 1] Operation not permitted: '/tmp/pip-fapb1E-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib'")]
```

carlescufi, 2016-08-04:
I'm a bit at a loss. Perhaps you could try uninstalling the current nrfutil version with pip and then installing it directly from pypi with `pip install nrfutil`? The version on pip is currently identical to master.

carlescufi, 2016-08-05:
We had someone internally with a similar error and uninstalling and installing again seemed to help. Did it work for you?

eliotstock, 2016-08-05:
I got it sorted, but it cost me about a day.

The issue is pip not respecting SIP (System Integrity Protection, new in Mac OS X 10.11, El Capitan) and is explained at length here: http://apple.stackexchange.com/a/210021/127382. I'm not sure why you guys haven't been getting the same problem, if you're on El Capitan.

What worked for me was to stick to the Mac OS X python version, NOT upgrade pip or setuptools explicitly (I think), then install nrfutil for my own user only, not for the whole system, using:

`pip install --user nrfutil`

I now have my nrfutil binary at `~/Library/Python/2.7/bin/nrfutil` and it works.

So, all good for master. I'm ready for SDK 12 when it comes. Now back to SDK 11 and the legacy image file format. I noticed there is no 0.5.x version in PyPI. If I install that from source instead using `python setup.py install`, it fails on permissions, trying to write to `/Library/Python/2.7/site-packages`. I don't want to install it there. But it doesn't let me use the `--install-dir` thing either. How can I install it where I want?

carlescufi, 2016-08-06:
Thanks for the feedback. We'll keep this in mind when solving similar issues.

About installing 0.5.1, you could simply run it from source?

I always run like this after cloning, from the root of the clone:

```
$ PYTHONPATH=$PWD python nordicsemi/__main__.py
```

eliotstock, 2016-08-06:
Oh, right. I never thought of that. That works. Thanks.

closed 2016-08-06 by eliotstock

carlescufi, 2016-08-07:
> I'm not sure why you guys haven't been getting the same problem, if you're on El Capitan.

I believe I did get the permissions issue, I'm running El Capitan. If I remember correctly I just changed those permissions on `/Library/Python/2.7/site-packages/` recursively to avoid the problem. Sorry I just realized about this now.

carlescufi, 2016-08-07:
Also for future reference, here's the full instructions on how to install pip with the stock OS X Python:

http://apple.stackexchange.com/a/223163