NordicSemiconductor/pc-nrfutil #40: Pythom3 request

hnhoan, 2016-12-12:
Please make it compatible with Pythom3.

bihanssen, 2016-12-16:
We have no immediate plans for python 3, but your request is noted.

bihanssen, 2017-03-22:
Closing the issue now, thought the request is noted. No new info is available at this point.

closed 2017-03-22 by bihanssen

FivazTimothee, 2017-09-26:
@bihanssen Hello Bjørn, I understand it's not in your current plans to switch to Python3 but could you sum up the main challenges that prevent you from doing so ? Are there some libraries only available for Python2 ?

bihanssen, 2017-09-26:
@FivazTimothee the main reason pc-ble-driver-py is based on Python2 is that we have internal users that cannot switch to Python3 because of library dependencies. 
We could add Python3 support, but that would have to come in addition to the Python2 support. It boils down to a matter of initial cost and maintenance cost.
We do want to add Py3 support the moment we do not have to keep the Py2 support.

michaldul, 2017-10-12:
@bihanssen  Are you open for contribution regarding this issue?

bihanssen, 2017-10-16:
@michaldul thanks for reaching out. We are open for contributions regarding this, though there's a catch. We won't be able to merge major PRs until a CLA has been signed, and the CLA system will be rolled out in November according to plans.

Also note that there is some activity regarding py3 in pc-ble-driver-py repo, see [pc-ble-driver-py issue #27](https://github.com/NordicSemiconductor/pc-ble-driver-py/issues/27).

nowox, 2017-11-30:
I am very sad that `nrfutils` is not Python3 compatible. Python2 is legacy and lots of big Python packages are now dropping support for Python2. Where will you make nrfutils compatible to Python3?