NordicSemiconductor/pc-nrfutil #46: DFU BLE returns "Fatal Python Error" on MacOS

solomonomeister, 2017-02-01:
Getting hard Python crashes when trying to run a DFU BLE command using nrfutil.

Sample command:
nrfutil dfu ble -ic NRF51 -pkg dfu_test_app_hrm_s132.zip -n "DfuTest"

Return:
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6

All versions of PIP, Python & nrfutil are up to date and freshly installed as of 02 Feb 2017. Running MacOS 10.12.3 Beta (16D30a).

Any ideas how to resolve?

bihanssen, 2017-03-16:
@solomonomeister do you still see this issue? Sorry for the late response.

solomonomeister, 2017-03-16:
@bihanssen if there haven't been any updates to nrfutil then yeah, its still happening.

bihanssen, 2017-03-17:
Ok, will try to reproduce the issue.

bihanssen, 2017-03-22:
I'm not able to reproduce the issue. Maybe you could provide some additional info:
1. Do you get the error message immediately after calling the command? Do you see any other output to the console?
2. What Python version are you using?

bihanssen, 2017-03-30:
Closing the issue due to can't reproduce. Please comment if there is more information available.

vikeri, 2017-05-11:
Getting the same issue still.

Python installed through homebrew.

macOS Sierra
Python 2.7.11
pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)

KSylwander, 2018-02-12:
Hello guys, 

I'm also getting this issue.

Python installed through Homebrew.

macOS HIgh Sierra
Python 2.7.14
pip 9.0.1

bihanssen, 2018-02-15:
Hi, thanks for reporting. Reopening the issue so we can check it out.
Does it happen every time you try to run it?

KSylwander, 2018-02-15:
Thank you.

As far as I know, yes.

spresley, 2018-04-13:
I am also seeing this issue, was anyone able to resolve this?

` nrfutil -v -v -v -v  dfu ble -ic NRF52 -f -pkg ble_app_buttonless_dfu_without_bonds_s132.zip
2018-04-13 12:05:58,688 Shared library: /Users/sapresley/.pyenv/versions/2.7.11/lib/python2.7/site-packages/pc_ble_driver_py/lib/macos_osx/libpc_ble_driver_shared_sd_api_v3.dylib
2018-04-13 12:05:58,688 Swig module name: pc_ble_driver_sd_api_v3
Fatal Python error: PyThreadState_Get: no current thread
[1]    53595 abort      nrfutil -v -v -v -v dfu ble -ic NRF52 -f -pkg`

mathiasap, 2018-04-24:
I was able to reproduce this issue on Python version 2.7.11 using pyenv. Nrfutil depends on a pc-ble-driver-py to communicate with CDC devices, but the the version you get by with "pip install pc_ble_driver_py" may not be compatible with Homebrew Python. The simplest way to work around this is by switching to system Python.
If you still want to use homebrew Python, you can build you own version of pc-ble-driver-py, and install it instead of the version in requirements.txt. Make sure that cmake is configured to use the correct libpython2.7.dylib for your version.

https://github.com/NordicSemiconductor/pc-ble-driver-py/

bihanssen, 2018-10-17:
Closing this issue as answered by https://github.com/NordicSemiconductor/pc-nrfutil/issues/46#issuecomment-383930818.

closed 2018-10-17 by bihanssen

ibeckermayer-neosensory, 2019-01-24:
@mathiasap I am running into a similar error, I was using virtualenv so decided to try your suggestion of switching back to system Python but that is giving me the same error. Any ideas? I don't quite understand what the problem is.

From the virtualenv I get:
```
[2019-01-23 19:31:52,273] [<module>:108] [INFO]: Shared library: /Users/IBM/.virtualenvs/ble_cli_serial/lib/python2.7/site-packages/pc_ble_driver_py/lib/macos_osx/libpc_ble_driver_shared_sd_api_v3.dylib
[2019-01-23 19:31:52,273] [<module>:109] [INFO]: Swig module name: pc_ble_driver_sd_api_v3
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```

From system python I get something similar:
```
[2019-01-23 19:32:52,336] [<module>:108] [INFO]: Shared library: /usr/local/lib/python2.7/site-packages/pc_ble_driver_py/lib/macos_osx/libpc_ble_driver_shared_sd_api_v3.dylib
[2019-01-23 19:32:52,336] [<module>:109] [INFO]: Swig module name: pc_ble_driver_sd_api_v3
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```

BaloMueller, 2019-07-12:
I can't fix it. Even by switching to system python on MacOS 10.14.5.

ojousima, 2019-12-19:
I'm having this issue as well on Mac Catalina, both on virtualenv / Python 2.7 and system Python 3.x

bihanssen, 2019-12-19:
@ojousima please try with a python.org release: https://www.python.org/downloads/.

ojousima, 2019-12-19:
After installing 3.8, here's what I get:
```
Otso-MacBook-Pro:src otsojousimaa$ nrfutil dfu ble -snr 682868184 -pkg targets/ruuvitag_b/armgcc/ruuvitag_b_armgcc_ruuvifw_default_v3.28.8_dfu_app.zip -ic NRF52 -n "Ruuvi"
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/nrfutil", line 11, in <module>
    load_entry_point('nrfutil==5.2.0', 'console_scripts', 'nrfutil')()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/nordicsemi/__main__.py", line 1106, in ble
    ble_driver_init(conn_ic_id)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/nordicsemi/__main__.py", line 70, in ble_driver_init
    from pc_ble_driver_py.ble_driver    import BLEDriver, Flasher
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pc_ble_driver_py/ble_driver.py", line 112, in <module>
    driver = importlib.import_module(SWIG_MODULE_NAME)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pc_ble_driver_py/lib/macos_osx/pc_ble_driver_sd_api_v3.py", line 28, in <module>
    _pc_ble_driver_sd_api_v3 = swig_import_helper()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pc_ble_driver_py/lib/macos_osx/pc_ble_driver_sd_api_v3.py", line 24, in swig_import_helper
    _mod = imp.load_module('_pc_ble_driver_sd_api_v3', fp, pathname, description)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: dynamic module does not define module export function (PyInit__pc_ble_driver_sd_api_v3)
```
Should I open another issue as this isn't anymore related to the original topic?