NordicSemiconductor/pc-nrfutil #48: DFU BLE failed on Ubuntu 16.04

junqingzou, 2017-02-10:
Hi,  this is Zou the Nordic Japan FAE reporting below issue.  

- JLink_Linux_V612i_x86_64
- nRF5x-Command-Line-Tools_9_3_1_Linux-x86_64
- nrfutil v2.1.0
Using nRF51 dongle as connectivity IC, firmware is as below
/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py/hex/sd_api_v2/connectivity_1.0.1_1m_with_s130_2.0.1.hex

>nrfutil dfu ble -ic NRF51 -pkg app_dfu_v1.1.zip -p /dev/ttyACM0 -n "DfuTarg"
OR
>nrfutil dfu ble -ic NRF51 -pkg app_dfu_v1.1.zip -p /dev/ttyACM0 -a DF:AF:6F:7A:0D:82

Result:
  [------------------------------------]    0%
Traceback (most recent call last):
  File "/usr/local/bin/nrfutil", line 11, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/__main__.py", line 660, in ble
    dfu.dfu_send_images()
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu.py", line 122, in dfu_send_images
    self._dfu_send_image(self.manifest.application)
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu.py", line 83, in _dfu_send_image
    self.dfu_transport.open()
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu_transport_ble.py", line 223, in open
    self.dfu_adapter.open()
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu_transport_ble.py", line 87, in open
    self.adapter.driver.open()
  File "/usr/local/lib/python2.7/dist-packages/wrapt/wrappers.py", line 561, in __call__
    args, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py/ble_driver.py", line 126, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13

Could you tell me what's wrong in above setup?
Please add Linux-environment DFU BLE test sample to "nrfutil_v1.1.pdf", too.

junqingzou, 2017-02-10:
Hi again, 

Issue is fixed if we let nrfutil to flash the connectivity firmware "automatically"
>nrfjprog -f NRF51 -e
>nrfutil dfu ble -ic NRF51 -pkg app_dfu_v1.1.zip -p /dev/ttyACM0 -n "DfuTarg" -f

Thanks!
BR, zou

bihanssen, 2017-02-13:
Hi, zou. Thanks for reporting.
The error you receive indicates that there is a problem communicating with the connectivity firmware.
I believe the problem here might be that you are using the 1Mbps variant of the connectivity firmware. nrfutil uses 115200 baudrate by default. Try to flash program the 115200 variant hex file and see if that resolves your issue.

junqingzou, 2017-02-13:
Thanks Bjorn for the details! 
Yes the "-f" option must have installed the 115200 version of firmware for me. 
I'll then tell customer to use 115200 instead of 1M if they want to program manually.

closed 2017-02-15 by bihanssen