NordicSemiconductor/pc-nrfutil #58: Is it possible to add nRF5x dfu with linux BLE driver?

ExCyb, 2017-03-20:
I have few puck.js devices (it is nRF52 based board) and want to update firmware. I have not any Android 6 device so I can not use Android tools (I've checked all phones I can access - nRF tools not supported).
At the same time, I have few linux boards with BLE supported. I can see my puck.js devices, for example:
` hcitool lescan
LE Scan ...
DC:0F:DF:1E:D4:73 DfuTarg
`
I'd want to update nRF firmware using the linux board. Is it possible?
Unfortunately, Python is not in my skills. I'm mostly low-level programmer (and so I can connect puck.js to JTAG/SWD and update firmware - but it seems to be ugly way).

bihanssen, 2017-03-21:
Hi, to use nrfutil for DFU you would need to do it through a nRF development kit, and I assume that is not possible with your linux board. We don't support DFU over linux bluez toolchain.
You could post a question on Devzone to hear if others have other ideas.

closed 2017-03-21 by bihanssen

wvengen, 2017-08-28:
That's too bad. Though [ota-dfu-python](https://github.com/astronomer80/ota-dfu-python) might work in some cases.