NordicSemiconductor/pc-nrfutil #69: Race condition in DfuTransportBle with debugger log enable

qung, 2017-05-02:
There is a race condition in DfuTransportBle with

mrodem, 2017-09-01:
After having examined the log file, I suspect that this may be related to #83.

First, we connect to device name "Nordic_Glucose" which, judging by the advertisement reports, has address 0xCE6D4432CD61:

```
hussar_test_utils.dfu_base: INFO: Connect to Nordic_Glucose
...
hussar_test_utils.dfu_base: DEBUG: Received advertisement report, address: 0xCE6D4432CD61, device_name: Nordic_Glucose
```

Next, I assume the target then enters DFU mode and changes its name to "CustomBL1" and updates its address to 0xCE6D4432CD62. We connect to this device to perform the DFU:

```
nordicsemi.dfu.dfu: INFO: Sending SoftDevice+Bootloader image.
nordicsemi.dfu.dfu_transport_ble: DEBUG: BLE: connect: target address: 0xNone
nordicsemi.dfu.dfu_transport_ble: INFO: BLE: Scanning...
nordicsemi.dfu.dfu_transport_ble: DEBUG: Received advertisement report, address: 0xCE6D4432CD62, device_name: CustomBL1
nordicsemi.dfu.dfu_transport_ble: INFO: BLE: Found target advertiser, address: 0xCE6D4432CD62, name: CustomBL1
nordicsemi.dfu.dfu_transport_ble: INFO: BLE: Connecting to 0xCE6D4432CD62
nordicsemi.dfu.dfu_transport_ble: DEBUG: Received advertisement report, address: 0xCE6D4432CD62, device_name:
nordicsemi.dfu.dfu_transport_ble: INFO: BLE: Found target advertiser, address: 0xCE6D4432CD62, name:
nordicsemi.dfu.dfu_transport_ble: INFO: BLE: Connecting to 0xCE6D4432CD62
```

We receive two advertisement reports for address 0xCE6D4432CD62: One with device name "CustomBL1" and one with device name "". The `target_device_name` in DfuAdapter is then overwritten and set to "": https://github.com/NordicSemiconductor/pc-nrfutil/blob/release/v3.2.0/nordicsemi/dfu/dfu_transport_ble.py#L177.

After the SoftDevice+Bootloader image is completed it reconnects, but now the `target_device_name` is "", so it ends up connecting to a different device:

```
nordicsemi.dfu.dfu_transport_ble: DEBUG: Received advertisement report, address: 0x08DF1FDAF7D7, device_name:
nordicsemi.dfu.dfu_transport_ble: INFO: BLE: Found target advertiser, address: 0x08DF1FDAF7D7, name:
nordicsemi.dfu.dfu_transport_ble: INFO: BLE: Connecting to 0x08DF1FDAF7D7
```

PR #88 should fix this issue.

mrodem, 2017-09-05:
Fix included in v3.2.1 release.

closed 2017-09-05 by mrodem