NordicSemiconductor/pc-nrfutil #75: DFU BLE, error after bootloader transferring

tdmmsky, 2017-05-17:
Hello,

I have issue to run nrfutil of version 2.3.0 with dfu ble feature.  

I compiled sources like boost (in 32 mode), pc-driver-ble-py(with 32 toolchain)

After compilation pc-driver-ble-py seems to work good, examples of advertisement working.
The main problem is with dfu ble feature. The device have bootloader from SDK 12.0 with custom fixing of using higher softdevice 3.1.0. The custom bootloader working properly with nrf connect desktop but not working with nrfutil ver 2.3 

There is possibility that app nrf connect can handle with older sdk bootloader and nrfutil not?
Below is output from nrfutil.

No target selected. Default device name: DfuTarg is used.
Please select connectivity serial port:
        0 : COM6 - 681706210
Enter your choice: : 0
Board already flashed with connectivity firmware.
  [------------------------------------]    0%
Traceback (most recent call last):
  File "nordicsemi\__main__.py", line 702, in <module>
  File "site-packages\click-6.7-py2.7.egg\click\core.py", line 722, in __call__
  File "site-packages\click-6.7-py2.7.egg\click\core.py", line 697, in main
  File "site-packages\click-6.7-py2.7.egg\click\core.py", line 1066, in invoke
  File "site-packages\click-6.7-py2.7.egg\click\core.py", line 1066, in invoke
  File "site-packages\click-6.7-py2.7.egg\click\core.py", line 895, in invoke
  File "site-packages\click-6.7-py2.7.egg\click\core.py", line 535, in invoke
  File "nordicsemi\__main__.py", line 688, in ble
  File "nordicsemi\dfu\dfu.py", line 122, in dfu_send_images
  File "nordicsemi\dfu\dfu.py", line 83, in _dfu_send_image
  File "nordicsemi\dfu\dfu_transport_ble.py", line 231, in open
  File "nordicsemi\dfu\dfu_transport_ble.py", line 336, in __set_prn
  File "nordicsemi\dfu\dfu_transport_ble.py", line 427, in __get_response
pc_ble_driver_py.exceptions.NordicSemiException: No Response: 0xFD
Failed to execute script __main__
f:\dd\vctools\crt\crtw32\stdcpp\thr\mutex.c(51): mutex destroyed while busy

mrodem, 2017-09-01:
Hi, sorry for the late response. This is difficult to investigate without having the bootloader you are using, and without seeing the exact nrfutil command you are attempting to run. Judging by the traceback, it seems to not receive any response when setting PRN. Does it work with other bootloaders, f.ex. the examples from SDK 13 or 14?

bihanssen, 2017-10-05:
Closing due to inactivity.

closed 2017-10-05 by bihanssen