NordicSemiconductor/pc-nrfutil #76: Installing version 0.5.2 does not install requirements

seppestas, 2017-05-24:
When installing nrfutil 0.5.2 using
```
pip install nrfutil==0.5.2
```
the requirements listed in [requirements.txt](https://github.com/NordicSemiconductor/pc-nrfutil/blob/v0.5.2/requirements.txt) do not get installed. This is because [setup.py](https://github.com/NordicSemiconductor/pc-nrfutil/blob/v0.5.2/setup.py) does not seem to install them.

Related to issue #65.

bencefr, 2017-05-30:
Until we provide a solution, could you install 2.3.0 first, then revert to 0.5.2, so requirements will be met?
```
pip install nrfutil==2.3.0
pip install nrfutil==0.5.2
```

seppestas, 2017-05-30:
Currently we just install the requirements directly, which seems to work:
```
pip install pyserial>=2.7 enum34>=1.0.4 click==5.1 ecdsa>=0.13
```

The method you described works as well, but is a bit less efficient since it requires nrfutil 2.3.0 to be downloaded, installed and uninstalled, and it is a bit less "maintainable" since the versions of the requirements required by later versions of nrfutil might be different from the versions required by nrfutil 0.5.2.

mrodem, 2017-07-25:
This issue was fixed by 799dd919, which was released with 0.5.3.

closed 2017-07-25 by mrodem

seppestas, 2017-07-25:
@mrodem @bihanssen Could you also release it on PyPI?

mrodem, 2017-07-25:
@seppestas Check now.

seppestas, 2017-07-25:
Works, thanks 👍