NordicSemiconductor/pc-nrfutil #83: dfu_transport_ble.py  could re-connect to incorrect BLE-device.

magnev, 2017-08-01:
When doing a DFU of a combined SD + APP image, after the SD update, nrfutil should re-connect to the bootloader to do the APP update. However, I've experienced cases where nrfutil re-connects to other BLE-devices nearby, and the DFU fails.
I've debugged and found the cause of this issue.

## When doing the SD update:
In the class DFUAdapter, when connect() is called, the parameters "target_device_name" and "target_device_addr" are given and stored in the class instance.
Upon receiving advertisement reports, the target_device values are compared to the BLE address and BLE device name of the advertising device. If one of them match nrfutil will initiate a connection.

However, DFUAdapter keeps getting advertisement reports until the connection is successfully completed. In some occasions it is possible to get advertisement packets without a BLE device name.
Here is a short log from a connection:

BLE: connect: target address: 0xNone, target name: DfuTest
BLE: Scanning...
Received advertisement report, address: 0xDCF79209A18C, device_name: DfuTest
BLE: Found target advertiser, address: 0xDCF79209A18C, name: DfuTest
BLE: Connecting to 0xDCF79209A18C
Received advertisement report, address: 0xDCF79209A18C, device_name: 
BLE: Found target advertiser, address: 0xDCF79209A18C, name: 
BLE: Connecting to 0xDCF79209A18C
BLE: Connected to <pc_ble_driver_py.ble_driver.BLEGapAddr object at 0x059929B0>
BLE: Connected to target
BLE: Service Discovery

From these lines of log and by looking at the code, we can now see that the value of self.target_device_name is an empty string.

## When doing the following APP update:
The empty string stored for self.target_device_name will be an issue when we try to re-connect to the bootloader after the SD update is succeeded.
This is because nrfutil will connect to a device with an empty BLE device name (empty string).

# FIX
In the on_gap_evt_adv_report() in DFUAdapter():
Remove the line:  self.target_device_name = dev_name

mrodem, 2017-08-31:
Thank you for reporting your findings.

I was struggling to figure out why we are overwriting `target_device_name` and `target_device_addr` in `on_gap_evt_adv_report()`. According to 13809768, we are doing this in case the name changes between connections. Let's say the user runs:

```
nrfutil dfu ble -ic NRF52 -pkg dfu_package.zip -p /dev/ttyACM0 -n "FooDevice"
```

Then, in the middle of the DFU, the target changes its name to "BarDevice". The user did not provide a target address, and now the name has changed, so the target can no longer be identified based on the information provided by the user. But because we got the `target_device_addr` from the advertisement report and returned it back to `DfuTransportBle`, we will still be able to identify the target device.

My conclusion then is that removing `self.target_device_name = dev_name` should be fine, as we still have the `target_device_addr`.

magnev, 2017-09-04:
Hi.

Yes, I came to the same conclusion.

mrodem, 2017-09-05:
Fix included in v3.2.1 release.

closed 2017-09-05 by mrodem