NordicSemiconductor/pc-nrfutil #91: Serial DFU error with nRF52840 DK and SDK 14 v5.0.0

JulesThuillier, 2017-09-18:
Hi ! I'm having a problem using the Serial DFU :/

Here is my setup :
- NRF52840 DK
- Softdevice SDK14 V5.0.0
- nrfutil v3.2.1
- OSX Sierra

I am using the **bootloader_secure_serial** from SDK14 examples

The public and private keys where set.

1. I **erase all** (*nrfjprog -f nrf52 --eraseall*)
2. I **flash the softdevice** (*nrfjprog -f nrf52 --program s140_nrf52840_5.0.0-2.alpha_softdevice.hex --sectorerase*)
3. I **flash the bootloader** (*nrfjprog/nrfjprog -f nrf52 --program nrf52840_bootloader_s140.hex --sectorerase*)
4. I connect my RTTViewer

**Here is what I see :**

*<info> nrf_sdh: Inside main
<debug> nrf_sdh: In nrf_bootloader_init
<debug> nrf_sdh: in weak nrf_dfu_init_user
<debug> nrf_sdh: In real nrf_dfu_init
<debug> nrf_sdh_soc: Running nrf_dfu_settings_init(sd_irq_initialized=false).
<debug> nrf_dfu_settings: Calling nrf_dfu_flash_init(sd_irq_initialized=false)...
<debug> nrf_dfu_settings: Initializing nrf_fstorage_nvmc backend.
<debug> nrf_sdh: Initializing the clock.
<debug> nrf_sdh: Enter nrf_dfu_continue
<error> nrf_sdh: Single: Invalid bank
<debug> nrf_sdh: Enter nrf_dfu_app_is_valid
<debug> nrf_sdh: Return false in valid app check
<debug> nrf_sdh: In nrf_dfu_transports_init
<debug> nrf_sdh: num transports: 1
<debug> nrf_sdh: UART initialized
<debug> nrf_sdh: After nrf_dfu_transports_init
<debug> nrf_dfu_settings: Calling nrf_dfu_flash_init(sd_irq_initialized=false)...
<debug> nrf_dfu_settings: Initializing nrf_fstorage_nvmc backend.
<debug> nrf_sdh: Waiting for events*

5. I **package** my application (*nrfutil pkg generate --application nrf52840_hid_keyboard.hex --key-file private.pem app_dfu_package.zip --debug-mode*)
6. And finally I try to **perform serial dfu** (*nrfutil dfu serial -pkg app_dfu_package.zip -p /dev/tty.usbmodem1421*)

**RTT Viewer**
<debug> nrf_sdh: Received ping 1
<debug> nrf_sdh: Sending Response: [0x9, 0x1]
<debug> nrf_sdh: Set receipt notif
<debug> nrf_sdh: Sending Response: [0x2, 0x1]
<debug> nrf_sdh: Received get serial mtu
<debug> nrf_sdh: Sending Response: [0x7, 0x1]

**And finally the error from nrfutil**
  [------------------------------------]    0%
Traceback (most recent call last):
  File "/usr/local/bin/nrfutil", line 9, in <module>
    load_entry_point('nrfutil==3.3.0', 'console_scripts', 'nrfutil')()
  File "/Library/Python/2.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/nrfutil-3.3.0-py2.7.egg/nordicsemi/__main__.py", line 655, in serial
    dfu.dfu_send_images()
  File "/Library/Python/2.7/site-packages/nrfutil-3.3.0-py2.7.egg/nordicsemi/dfu/dfu.py", line 122, in dfu_send_images
    self._dfu_send_image(self.manifest.application)
  File "/Library/Python/2.7/site-packages/nrfutil-3.3.0-py2.7.egg/nordicsemi/dfu/dfu.py", line 90, in _dfu_send_image
    self.dfu_transport.send_init_packet(data)
  File "/Library/Python/2.7/site-packages/nrfutil-3.3.0-py2.7.egg/nordicsemi/dfu/dfu_transport_serial.py", line 239, in send_init_packet
    self.__stream_data(data=init_packet)
  File "/Library/Python/2.7/site-packages/nrfutil-3.3.0-py2.7.egg/nordicsemi/dfu/dfu_transport_serial.py", line 405, in __stream_data
    response = self.__calculate_checksum()
  File "/Library/Python/2.7/site-packages/nrfutil-3.3.0-py2.7.egg/nordicsemi/dfu/dfu_transport_serial.py", line 347, in __calculate_checksum
    (offset, crc) = struct.unpack('<II', bytearray(response))
TypeError: 'NoneType' object is not iterable

bihanssen, 2017-09-20:
Hi, thanks for reporting.
We have looked into the issue but we are not able to reproduce it. 
What we tested: bootloader and ble_hids_keyboard app compiled with GCC version.
nrfutil from pip, versions 2.3.0, 3.0.0, 3.2.0 and 3.3.0 all seem to work.

The error you get indicates that the device does not respond to a checksum command, but we are as I said not able to reproduce it.

idelsink, 2017-09-20:
@bihanssen Are you compiling the project/bootloader with or without optimization?

JulesThuillier, 2017-09-21:
@idelsink I am using the Makefile provided for armgcc with the bootloader serial project, and I check here are the optimization flags being used : **OPT = -Os -g3**

@bihanssen Thanks for checking ! I have just tried another dev board just in case but same problem here :/ I am out of ideas right now. Should I try with the projects from SDK 13 ? 

One thing I just found out is that I cannot program the softdevice and then the bootloader if I don't use the "sectorerase" option, which seems wring as those two hex shouldn't overlap :

nrfjprog -f nrf52 --eraseall
nrfjprog -f nrf52 --program s140_nrf52840_5.0.0-2.alpha_softdevice.hex 
nrfjprog -f nrf52 --program nrf52840_bootloader_s140.hex
*Parsing hex file.
Reading flash area to program to guarantee it is erased.
ERROR: The area to write is not erased.*

Maybe those are two different problems, but just in case I preferred to share this information.

Thanks for your help :)

bihanssen, 2017-09-21:
@JulesThuillier I`ll have to get back to you next week om this.

JulesThuillier, 2017-09-22:
I tried again, using the bootloader settings this time (*nrfutil settings generate --family NRF52840 --application ble_app_hids_keyboard_pca10056_s140.hex --application-version 3 --bootloader-version 1 --bl-settings-version 1 settings.hex*), and a new key.

This time in the RTT Viewer it goes there :
*<debug> nrf_sdh: Sending Response: [0x9, 0x1]
<debug> nrf_sdh: Set receipt notif
<debug> nrf_sdh: Sending Response: [0x2, 0x1]
<debug> nrf_sdh: Received get serial mtu
<debug> nrf_sdh: Sending Response: [0x7, 0x1]
<debug> nrf_sdh: Received select object
<info> nrf_dfu_flash: Valid Command: NRF_DFU_OBJECT_OP_SELECT
<debug> nrf_sdh: Sending Response: [0x6, 0x1]*

A little further....before getting the same issue.

Another thing I found out it that if I program the softdevice, bootloader (merged with settings) and the application, it doesn't start the app and stays stuck in the bootloader :
*
<info> nrf_sdh: Inside main
<debug> nrf_sdh: In nrf_bootloader_init
<debug> nrf_sdh: in weak nrf_dfu_init_user
<debug> nrf_sdh: In real nrf_dfu_init
<debug> nrf_sdh_soc: Running nrf_dfu_settings_init(sd_irq_initialized=false).
<debug> nrf_dfu_settings: Calling nrf_dfu_flash_init(sd_irq_initialized=false)...
<debug> nrf_dfu_settings: Initializing nrf_fstorage_nvmc backend.
<debug> nrf_sdh: Initializing the clock.
<debug> nrf_sdh: Enter nrf_dfu_continue
<debug> nrf_sdh: Valid App
<debug> nrf_sdh: Enter nrf_dfu_app_is_valid
<debug> nrf_sdh: Return false in CRC
<debug> nrf_sdh: In nrf_dfu_transports_init
<debug> nrf_sdh: num transports: 1
<debug> nrf_sdh: UART initialized
<debug> nrf_sdh: After nrf_dfu_transports_init
<debug> nrf_dfu_settings: Calling nrf_dfu_flash_init(sd_irq_initialized=false)...
<debug> nrf_dfu_settings: Initializing nrf_fstorage_nvmc backend.
<debug> nrf_sdh: Waiting for events
*

Is it possible that the **--bl-settings-version 1** should be different than 1 ? There is very little info on the expected value. My package is generate with **--debug-mode** so there shouldn't be any problem validating and going to the app.

bihanssen, 2017-10-05:
@JulesThuillier I'm not sure what the problem here is, sorry. Could you please try to post on DevZone, You might get better response there.

IvanSanchez, 2017-10-06:
Can reproduce.

Actually I can see a few different error messages from the Python end:

```
[snip]
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu.py", line 83, in _dfu_send_image
    self.dfu_transport.open()
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu_transport_serial.py", line 190, in open
    if self.__ping() == False:
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu_transport_serial.py", line 321, in __ping
    + 'Expected: 0x{:02X} Received: 0x{:02X}'.format(operation, resp[1]))
NameError: global name 'operation' is not defined
```

```
[snip]
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu.py", line 90, in _dfu_send_image
    self.dfu_transport.send_init_packet(data)
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu_transport_serial.py", line 242, in send_init_packet
    self.__stream_data(data=init_packet)
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu_transport_serial.py", line 405, in __stream_data
    response = self.__calculate_checksum()
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/dfu/dfu_transport_serial.py", line 347, in __calculate_checksum
    (offset, crc) = struct.unpack('<II', bytearray(response))
TypeError: 'NoneType' object is not iterable
```

This is reproducible consistently in my Debian box. I'm also using the .hex files from the SDK v14 for the softdevice and serial bootloader, but on a nRF52832 devkit instead of a 52840.

weizhen1883, 2018-02-21:
have you figure out it yet? I meet the same problem.

mrodem, 2018-03-06:
I have done some testing on this, and I believe this is related to MSD being enabled on the devkit. See my response on https://devzone.nordicsemi.com/f/nordic-q-a/30742/nrfutil-dfu-serial-return-typeerror-nonetype-object-is-not-iterable.

In addition, improved error output has been added in #130.

closed 2018-03-06 by mrodem

jefffhaynes, 2019-02-05:
I have this same problem and I am not using MSD. Not entirely sure why this was closed with no resolution...

rebeccasc, 2021-07-22:
I still have a quite similar looking error while trying to flash an application via uart (NRF52832, PCA10040, nRF5_SDK 16, nrfutil version 6.1.0). The bootloader works fine (it's the example bootloader: secure_bootloader_pca10040_uart) and the device is in bootloader mode while trying to flash the blinky example with `nrfutil dfu usb-serial -pkg blinky.zip -p COM3 -b 115200`. Also tried it with `--debug-mode` flag.
My exception looks like this:
```
  [------------------------------------]    0%
Traceback (most recent call last):
  File "C:\Python\Python38\Scripts\nrfutil-script.py", line 33, in <module>
    sys.exit(load_entry_point('nrfutil==6.1.0', 'console_scripts', 'nrfutil')())
  File "c:\python\python38\lib\site-packages\click\core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "c:\python\python38\lib\site-packages\click\core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "c:\python\python38\lib\site-packages\click\core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python\python38\lib\site-packages\click\core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python\python38\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python\python38\lib\site-packages\click\core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "c:\python\python38\lib\site-packages\nordicsemi\__main__.py", line 1014, in usb_serial
    do_serial(package, port, connect_delay, flow_control, packet_receipt_notification, baud_rate, serial_number, False,
  File "c:\python\python38\lib\site-packages\nordicsemi\__main__.py", line 970, in do_serial
    dfu.dfu_send_images()
  File "c:\python\python38\lib\site-packages\nordicsemi\dfu\dfu.py", line 127, in dfu_send_images
    self._dfu_send_image(self.manifest.application)
  File "c:\python\python38\lib\site-packages\nordicsemi\dfu\dfu.py", line 88, in _dfu_send_image
    self.dfu_transport.open()
  File "c:\python\python38\lib\site-packages\nordicsemi\dfu\dfu_transport_serial.py", line 217, in open
    self.__get_mtu()
  File "c:\python\python38\lib\site-packages\nordicsemi\dfu\dfu_transport_serial.py", line 366, in __get_mtu
    self.mtu = struct.unpack('<H', bytearray(response))[0]
TypeError: cannot convert 'NoneType' object to bytearray
```

I also tried @mrodem's hint to disable MSD but it didn't help. 
Does anyone have an idea what the problem could be?