NordicSemiconductor/pc-nrfutil #100: Hangs on some app files

lcrocker, 2017-11-04:
Some app files (sample attached) cause "nrfutil settings generate" to go into an infinite loop. Can nearly even crash Linux--machine runs slowly, and running "top" in another terminal shows nrfutil using all of ram.

I suspect it's because this hex file contains a flash section at 0x23000 and a ram section at 0x20002180. But even if it's naively allocating a single large buffer, that's only half a gig (is my math right on that?). Doesn't seem like that should stress my 16GB box.

[base.hex.gz](https://github.com/NordicSemiconductor/pc-nrfutil/files/1442763/base.hex.gz)

IvanSanchez, 2017-11-13:
Keep in mind that the DFU procedure (as currently implemented) doesn't allow updating the contents of the volatile RAM (or of a QSPI peripheral, for that matter). Check the [infocenter documentation on DFU banking](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v14.1.0%2Flib_bootloader_dfu_banks.html) for some context. (And, yes, your math is right, the [RAM addresses start at 0x20000000 ~= 0.5 GiB](https://developer.arm.com/docs/ddi0439/latest/programmers-model/system-address-map)).

Looking at the code, I'll guess that a ""proper"" fix is to limit the size of the UICR to be probed, as right now there is a loop from the UICR start until the highest address of the parsed .hex file:

https://github.com/NordicSemiconductor/pc-nrfutil/blob/05e178b421c465210f1a742dc8110a0422d27f46/nordicsemi/dfu/nrfhex.py#L94

Looks like the code needs a better way to clamp the keys in the `_buf` dictionary.

(Handling of other address ranges could be improved, but let's leave that for later)

bihanssen, 2019-10-04:
Closing as won't fix. Hex files for the supported memory range would cause the reported behaviour.

closed 2019-10-04 by bihanssen