NordicSemiconductor/pc-nrfutil #103: PROP_IPv6_ICMP_PING_OFFLOAD error on Thread DFU

IvanSanchez, 2017-11-13:
Running `nrfutil dfu thread -pkg /tmp/Thread_DFU-test/app_dfu_package.zip --port /dev/ttyACM0  -f` in my Debian box fails **consistently** with the following error stack:

```
Address not specified. Using ff03::1 (all Thread nodes)
Using connectivity board at serial port: /dev/ttyACM0
Board already flashed with connectivity firmware.
Opening serial to /dev/ttyACM0 @ 115200
2017-11-10 16:55:45,473 Failed to set property PROP_IPv6_ICMP_PING_OFFLOAD
Traceback (most recent call last):
  File "/home/ivan/devel/pc-nrfutil/nordicsemi/__main__.py", line 918, in thread
    transport.open()
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/thread/tncp.py", line 211, in open
    if (not self._attach_to_network()):
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/thread/tncp.py", line 102, in _attach_to_network
    self._set_property(*props)
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/thread/tncp.py", line 87, in _set_property
    raise Exception("Failed to set property {}".format(self.__class__._propid_to_str(propid)))
Exception: Failed to set property PROP_IPv6_ICMP_PING_OFFLOAD
```

On one hand, I'd like the DFU procedure to just succeed. All my attempts to run `nrfutil dfu thread -f` on Debian fail with this error

On the other hand, as a user, I'd like to see a more meaningful error message - what is `PROP_IPv6_ICMP_PING_OFFLOAD` and why should I care about it, what should I do to fix that?

Is this error caused by some failure at initializing something more general (e.g. sending the first IP packet)? If there is a root cause for this, I'd like to see it in the error message.

wbober, 2017-11-14:
@IvanSanchez 
Unfortunately, the issue is not that consistent elsewhere. This typically shows up after flashing the firmware but doing a hard reset after flashing should make it disappear.

bihanssen, 2019-10-18:
Closing as won't fix

closed 2019-10-18 by bihanssen

nikhil30081995, 2021-09-07:
Did anyone found any solution to this  ?