NordicSemiconductor/pc-nrfutil #104: "Failed to reset NCP" error on Thread DFU

IvanSanchez, 2017-11-13:
Running `nrfutil dfu thread -pkg /tmp/Thread_DFU-test/app_dfu_package.zip --port /dev/ttyACM0` (without `-f`) in my Debian box fails **consistently** with the following error stack:

```
Traceback (most recent call last):
  File "/home/ivan/devel/pc-nrfutil/nordicsemi/__main__.py", line 918, in thread
    transport.open()
  File "/usr/local/lib/python2.7/dist-packages/nordicsemi/thread/tncp.py", line 208, in open
    raise Exception('Failed to reset NCP. Please flash connectvity firmware.')
Exception: Failed to reset NCP. Please flash connectvity firmware.
```

This is particularly annoying, since I did run a `nrfutil dfu thread -f ...` just a few seconds earlier ([just to find a different error](#103)).

Is there any way to check the serialized softdevice messages and check what's going on? Reflashing the connectivity firmware obviously doesn't help.

wbober, 2017-11-14:
@IvanSanchez So to reiterate:
You've flashed connectivity firmware by calling ```nrfutil``` with ```-f``` and got ``` PROP_IPv6_ICMP_PING_OFFLOAD```.
Without resetting the board in any way (hard or soft), you have invoked ```nrfutil``` without ```-f``` to get 
```Failed to reset NCP. Please flash connectvity firmware.```.

If so then this issue is essentially a duplicate of #103. The board is not responding to commands after it has been flashed with the connectivity firmware. Since we don't compare the contents of flash with the connectivity firmware we don't have any means to discriminate between incorrect firmware and a board which doesn't reply to commands. Hence the message.

IvanSanchez, 2017-11-14:
> Since we don't compare the contents of flash with the connectivity firmware

Is there any downside to doing so?

I mean, the current implementation looks like:

```
def thread(...):
    # ...
    if flasher.fw_check():
        click.echo("OK.")
    if flash_connectivity:
        flasher = NCPFlasher(serial_port=port, snr = jlink_snr)
        else:
            flasher.fw_flash()
            click.echo("Connectivity firmware flashed.")
    
    dfu.trigger(...)
```

What if, instead of that, we do something more like...

```
def thread(...):
    # ...
    
    flasher = NCPFlasher(serial_port=port, snr = jlink_snr)
    if flasher.fw_check():
        click.echo("Board already flashed with connectivity firmware.")
    else:
        if flash_connectivity:
            flasher.fw_flash()
            click.echo("Connectivity firmware flashed.")
        else:
            click.echo("Board is not flashed with connectivity firmware. Try again, using the -f option.")
            sys.exit(-1)
    dfu.trigger(...)
```

This way we could check independently whether we got a connectivity IC, and whether the connectivity IC is responding to reset commands.

wbober, 2017-11-14:
@IvanSanchez I believe it takes a bit of time to do the check, what is a bit annoying when you do the development. But I guess we could add a ```skip-check``` flag.

bihanssen, 2019-10-18:
Closing as won't fix.

closed 2019-10-18 by bihanssen