NordicSemiconductor/pc-nrfutil #106: Verbose IPv6 addresses in thread DFU output lack context

IvanSanchez, 2017-11-13:
Running `nrfutil -v dfu thread -pkg ...\app_dfu_package.zip -p COM10` (note `-v`) produces the following output:

```
2017-11-13 17:07:50,131 Done
2017-11-13 17:07:50,148 fdde:ad00:beef::ff:fe00:fc00
2017-11-13 17:07:50,148 fdde:ad00:beef::ff:fe00:a000
2017-11-13 17:07:50,148 fdde:ad00:beef:0:c7f8:61c6:baf9:f2b0
2017-11-13 17:07:50,148 fe80::30c3:b84:1071:6d93
2017-11-13 17:07:56,148 Triggering DFU on Endpoint(addr=IPv6Address(u'ff03::1'), port=5683)
```

What are those IPv6 addresses? Are they addresses in my host computer? In the NCP? Are they addresses of DFU targets? Multicast groups?

Right now this verbose output is more confusing than helpful. Please add context as to what those IPv6 addresses represent.

rlubos, 2017-11-14:
These IPv6 adresses are addresses that are assigned to the NCP in the Thread network. Will update the log to be more verbose about it.

bihanssen, 2019-10-18:
Closing as fixed.

closed 2019-10-18 by bihanssen