NordicSemiconductor/pc-nrfutil #107: Thread DFU should gracefully timeout

IvanSanchez, 2017-11-13:
Running `nrfutil -v dfu thread -pkg ...\app_dfu_package.zip -p COM10` (note `-v`) produces the following output:

```
2017-11-13 17:07:56,148 Sending message to ff03::1:5683
2017-11-13 17:07:56,164 Message 'R\x02na\xac\x03\xb1t\xff\x10\x00\x00\x00\x86\x02\x12\xebs\x00\x00&d*\xb5kj' sent successfully
2017-11-13 17:07:56,164 Sending request - Token: ac03, Host: ff03::1, Port: 5683
Press <ENTER> to terminate
2017-11-13 17:07:58,171 Request timed out
2017-11-13 17:07:58,171 Exchange removed, Message ID: 28257.
2017-11-13 17:07:58,171 Sending message to ff03::1:5683
2017-11-13 17:07:58,187 Message 'R\x02na\xac\x03\xb1t\xff\x10\x00\x00\x00\x86\x02\x12\xebs\x00\x00&d*\xb5kj' sent successfully
2017-11-13 17:07:58,187 Sending request - Token: ac03, Host: ff03::1, Port: 5683
2017-11-13 17:08:00,194 Request timed out
2017-11-13 17:08:00,194 Exchange removed, Message ID: 28257.
2017-11-13 17:08:00,194 Sending message to ff03::1:5683
2017-11-13 17:08:00,210 Message 'R\x02na\xac\x03\xb1t\xff\x10\x00\x00\x00\x86\x02\x12\xebs\x00\x00&d*\xb5kj' sent successfully
2017-11-13 17:08:00,210 Sending request - Token: ac03, Host: ff03::1, Port: 5683
2017-11-13 17:08:02,213 Request timed out
2017-11-13 17:08:02,213 Exchange removed, Message ID: 28257.
```

...and then nothing happens.

It's obvious that something waiting two seconds to try something three times, and then hanging.

Why three times? Why keep the program hanging out after that time? An improvement would be to keep repeating the action indefinitely, or exit the program after 3 failures. Even better: provide an extra parameter to the user, letting they configure how many retries to do (and something like "0 means infinite").

wbober, 2017-11-14:
@IvanSanchez The program is not hanging it's waiting for requests. 

There are three ways that nrfutil works in case of Thread DFU:
1) if destination address is not specified: it acts as CoAP Server and waits for firmware requests. This is essentially what you are seeing in the log. After starting it multicasts a trigger message ("I'm here") to all nodes to inform the network of it existence. After that it waits for firmware download requests. 
2) if unicast address is specified: it acts as a CoAP server and triggers firmware upgrade on a single node. 
3) if multicast address is specified: it acts as a CoAP client and disseminates firmware to multiple nodes at once

The number of retries is arbitrary but it's not critical. In a real world scenario a device will periodically query for DFU servers. Missing the initial trigger will only postpone the download.

I object adding infinite number of retries. It's not how this is intended to work. If there is something that could be improved here is to add an option to daemonize the server.

bihanssen, 2019-10-18:
Closing as won't fix.

closed 2019-10-18 by bihanssen