NordicSemiconductor/pc-nrfutil #113: how generate zip file with legacy version(0.5.2)

fkeujjpdc, 2017-11-17:
hello:
i wan't convert hex file to zip file without key-file . but i use "pip install  nfrutil" download new version(1.5.0)
that it must use pem file . how can i generate zip file without key-file.

ArthurHCL, 2017-12-19:
I also want to know the issue,for help......

svenKautlenbach, 2018-02-02:
checkout branch 0_5_3 or by tag v0.5.3 that has the legacy support

bihanssen, 2018-10-17:
The response of @svenKautlenbach is correct, you can check out the branch or tag and build from there.

In theory it should be possible to install nrfutil 0.5.3 with pip, with the command `pip install nrfutil==0.5.3`, though I'm having problems that seem to be related to incompatiblilties between pip versions.

closed 2018-10-17 by bihanssen