NordicSemiconductor/pc-nrfutil #114: nrfutil pkg generate.... MemoryError

Boisi73, 2017-11-22:
Hi, 
I'm having a problem using the nrfutil to generate a package ZIP.

My setup:
NRF52
Softdevice SDK14 V5.0.0
nrfutil 3.3.2
GCC

I try to pack my app.hex file like: 
nrfutil pkg generate --hw-version 52 --application-version 1 --application app.hex --sd-req 0x9d --key-file private.key app_dfu_br.zip

..and this error appears:
Traceback (most recent call last):
  File "nordicsemi\__main__.py", line 871, in <module>
  File "site-packages\click\core.py", line 722, in __call__
  File "site-packages\click\core.py", line 697, in main
  File "site-packages\click\core.py", line 1066, in invoke
  File "site-packages\click\core.py", line 1066, in invoke
  File "site-packages\click\core.py", line 895, in invoke
  File "site-packages\click\core.py", line 535, in invoke
  File "nordicsemi\__main__.py", line 580, in generate
  File "nordicsemi\dfu\package.py", line 339, in generate_package
  File "nordicsemi\dfu\package.py", line 489, in normalize_firmware_to_bin
  File "nordicsemi\dfu\nrfhex.py", line 80, in __init__
  File "nordicsemi\dfu\nrfhex.py", line 94, in _removeuicr
MemoryError
Failed to execute script __main__

Some HEX Files are packed and some are not.

IvanSanchez, 2017-11-22:
Can you provide the .hex file that makes this fail? What's the highest address referenced by that hex file?

Boisi73, 2017-11-22:
Hi,

here the HEX File

IvanSanchez, 2017-11-22:
Your .hex file contains data at offset `0x2000241c`, which means you're trying to fill RAM with some initial values.

The DFU procedure only updates non-volatile memory; it cannot update RAM. See http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v14.1.0%2Flib_bootloader_dfu_banks.html for the technical details, notice how the DFU procedure only affects non-volatile memory.

Work around this issue by making sure that your code doesn't create a .hex file that sets up initial values in RAM.

The fact that `nrfutil` has problems when dealing with .hex files containing data in very high addresses is known, so I'm closing this as a duplicate of #100.

closed 2017-11-22 by IvanSanchez