NordicSemiconductor/pc-nrfutil #118: Pyinstaller nrfutil.spec on mac fails with ValueError: Unknown Mach-O header

fodra, 2017-12-04:
I am able to install nrfutil using `python setup.py install` on Mac.

But I'd like to compile it and build an executable file on Mac.
I used `pyinstaller ./nrfutil.spec` but it returns an error:

    ValueError: Unknown Mach-O header: 0x4275696c in <open file '/Users/andrew/Library/Application Support/pyinstaller/bincache00_py27_64bit/lib/win/x86_32/build_info.txt', mode 'rb' at 0x109e89930>

I'm wondering if this is possible on Mac. Full stack trace is:

`
Traceback (most recent call last):
  File "/Users/andrew/Envs/nrfutil/bin/pyinstaller", line 11, in <module>
    sys.exit(run())
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/__main__.py", line 92, in run
    run_build(pyi_config, spec_file, **vars(args))
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/__main__.py", line 46, in run_build
    PyInstaller.building.build_main.main(pyi_config, spec_file, **kwargs)
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/build_main.py", line 791, in main
    build(specfile, kw.get('distpath'), kw.get('workpath'), kw.get('clean_build'))
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/build_main.py", line 737, in build
    exec(text, spec_namespace)
  File "<string>", line 37, in <module>
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/api.py", line 411, in __init__
    strip_binaries=self.strip, upx_binaries=self.upx,
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/api.py", line 196, in __init__
    self.__postinit__()
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/datastruct.py", line 161, in __postinit__
    self.assemble()
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/api.py", line 259, in assemble
    dist_nm=inm)
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/utils.py", line 207, in checkCache
    dylib.mac_set_relative_dylib_deps(cachedfile, dist_nm)
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/PyInstaller/depend/dylib.py", line 308, in mac_set_relative_dylib_deps
    dll = MachO(libname)
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/macholib/MachO.py", line 95, in __init__
    self.load(fp)
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/macholib/MachO.py", line 110, in load
    self.load_header(fh, 0, size)
  File "/Users/andrew/Envs/nrfutil/lib/python2.7/site-packages/macholib/MachO.py", line 147, in load_header
    header, fh))
ValueError: Unknown Mach-O header: 0x4275696c in <open file '/Users/andrew/Library/Application Support/pyinstaller/bincache00_py27_64bit/lib/win/x86_32/build_info.txt', mode 'rb' at 0x109e89930>
`

fodra, 2017-12-04:
could I just use the installed bin? found here?
/Users/andrew/Envs/nrfutil/bin/nrfutil

bihanssen, 2018-10-17:
@fodra sorry for late response. I have not seen that error before. If this is still an issue, please repost on [DevZone](http://devzone.nordicsemi.com).

closed 2018-10-17 by bihanssen