NordicSemiconductor/pc-nrfutil #125: Error code 13

rpelc, 2018-02-05:
Hello,
I have problem during dfu otau over ble connection.
I use linux version of the nrfutil - my OS version is Linux-4.9.0-3-amd64-x86_64-with-debian-9.1
In some cases i get error code 13 when i try to start the ota update.
It happens randomly, to remove it i have to reset the dongle i'm using to update.
Can you tell my what this error is about and how to avoid it?

bihanssen, 2018-02-27:
Hi, @rpelc
error code 13 means there was a timeout during the opening of the transport protocol. This may be related to issues that have been addressed in the lower level pc-ble-driver (see [change logs](https://github.com/NordicSemiconductor/pc-ble-driver/releases). These fixes have not found their way to nrfutil yet since the reference to pc-ble-driver has not been updated.

What you may try for now is to use the connectivity hex files from [v2.2.1](https://github.com/NordicSemiconductor/pc-ble-driver/tree/v2.2.1/hex) instead, and see if that helps.

bihanssen, 2018-10-17:
The error 13 in nrfutil is originated in pc-ble-driver and/or pc-ble-driver-py. A new version of pc-ble-driver is imminent, and after that a new version of pc-ble-driver-py. I would recommend to wait for those updates as they will probably improve stability also for this issue.
If further questions, please repost on [DevZone](http://devzone.nordicsemi.com).

closed 2018-10-17 by bihanssen