NordicSemiconductor/pc-nrfutil #133: Not able to load protobuf DLL

andvib, 2018-03-08:
Whenever running nrfutil with any option, it throws an ImportError that says it's not able to load the protobuf DLL:

Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\nrfutil.exe\__main__.py", line 5, in <module>
  File "c:\python27\lib\site-packages\nordicsemi\__main__.py", line 49, in <module>
    from nordicsemi.dfu.bl_dfu_sett import BLDFUSettings
  File "c:\python27\lib\site-packages\nordicsemi\dfu\bl_dfu_sett.py", line 53, in <module>
    from nordicsemi.dfu.package import Package
  File "c:\python27\lib\site-packages\nordicsemi\dfu\package.py", line 53, in <module>
    from nordicsemi.dfu.init_packet_pb import *
  File "c:\python27\lib\site-packages\nordicsemi\dfu\init_packet_pb.py", line 37, in <module>
    import dfu_cc_pb2 as pb
  File "c:\python27\lib\site-packages\nordicsemi\dfu\dfu_cc_pb2.py", line 43, in <module>
    from google.protobuf import descriptor as _descriptor
  File "c:\python27\lib\site-packages\google\protobuf\descriptor.py", line 46, in <module>
    from google.protobuf.pyext import _message
ImportError: DLL load failed: The specified procedure could not be found.

This is for Python 2.7.11, and happens with both the 3.4 and 3.5 versions of nrfutil and installing nrfutil via pip. Pip says that protobuf is already installed. When only calling 'nrfutil' it displays the info as it should, but when adding any option it raises the error.

bihanssen, 2018-03-12:
Hi,
it seems there is a problem with nrfutil and the latest version of protobuf. Whether the problem is due to protobuf itself, or how nrfutil uses it, is unclear.
A temporary workaround that seems to work is to downgrade protobuf to the previous version, 3.5.1:
`pip install protobuf==3.5.1`

bihanssen, 2018-03-19:
Update: there is now available a new version of protobuf: 3.5.2.post1 that also seems to fix the problem. @andvib can you confirm if upgrading to the latest available version solves the problem you're seeing?

andvib, 2018-03-20:
Yes, using protobuf 3.5.2.post1 fixes my issue

bihanssen, 2018-03-20:
Thanks for reporting. Closing the issue as fixed.

Bloodroxxor, 2018-09-03:
Exact same error happening for me again with 3.6.1
It is not happening with 3.6.0 

Python 3.6.0 with Anaconda 4.3.1 under windows

bihanssen, 2018-09-05:
@Bloodroxxor thanks for reporting. Reopening the issue.

Oswfrans, 2018-09-12:
> Exact same error happening for me again with 3.6.1
> It is not happening with 3.6.0
> 
> Python 3.6.0 with Anaconda 4.3.1 under windows

This worked for me thank you for reporting

NavyaSandrana, 2018-10-13:
TypeError: __init__() got an unexpected keyword argument 'serialized_options'

I am getting this error if i import tensorflow as 
python -c "import tensorflow as tf; print(tf.__version__)" 

Could you please help me in solving the error..

dhananjayharel, 2018-10-24:
> TypeError: **init**() got an unexpected keyword argument 'serialized_options'
> 
> I am getting this error if i import tensorflow as
> python -c "import tensorflow as tf; print(tf.**version**)"
> 
> Could you please help me in solving the error..

I am also facing the same issue. after downgrading protobuf to 3.5.2
please suggest the solution.

dhananjayharel, 2018-10-24:
Please check the python version.
It works well with python version 3.6.1
My previous version was 3.6.0. I have upgraded to 3.6.1 and its working fin now.
No need to downgrade the protobuf.