NordicSemiconductor/pc-nrfutil #147: nrfutil dfu ble on Ubuntu 18.04 does not complete

pabigot, 2018-05-25:
I'm trying to verify DFU update capability in this configuration:
* Ubuntu 18.04 64-bit host
* nRF5-SDK 12.3.0 examples/dfu/bootloader_secure on target with S130 2.0.1
* nrfutil 3.5.1 with connectivity_2.0.1_115k2_with_s130_2.0.1.hex installed on the programmer

Invoking with:

    nrfutil dfu ble \
     -ic NRF51 \
     -pkg /tmp/app_dfu_package.zip \
     -p /dev/ttyUSB0 \
     -a E2:82:FE:1D:97:AC \
     -n DfuTarg

the debug output on the target shows the connection is made, and data starts being stored, but then the programmer terminates with:

```
  File "/home/pab/.local/lib/python2.7/site-packages/nordicsemi/__main__.py", line 838, in ble
    dfu.dfu_send_images()
  File "/home/pab/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu.py", line 129, in dfu_send_images
    self._dfu_send_image(self.manifest.application)
  File "/home/pab/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu.py", line 102, in _dfu_send_image
    self.dfu_transport.send_firmware(data)
  File "/home/pab/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_ble.py", line 549, in send_firmware
    response['crc'] = self.__stream_data(data=data, crc=response['crc'], offset=i)
  File "/home/pab/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_ble.py", line 619, in __stream_data
    self.dfu_adapter.write_data_point(map(ord, to_transmit))
  File "/home/pab/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_ble.py", line 332, in write_data_point
    self.adapter.write_cmd(self.conn_handle, DFUAdapter.DP_UUID, data)
  File "/home/pab/.local/lib/python2.7/site-packages/pc_ble_driver_py/ble_adapter.py", line 366, in write_cmd
    raise e
pc_ble_driver_py.exceptions.NordicSemiException: Failed to ble_gattc_write. Error code: 3
```

Failure generally occurs after one or two, sometimes as many as 9, successful 256 By transfers.  The maximum I've been able to successfully transfer is 16 KiBy.  After failure I have to reset the target board to try again.

My hypothesis is that `3` is `NRF_ERROR_INTERNAL` and indicates a checksum error transferring data from the Linux host into the programmer.  I'm working to verify that, but hope somebody might provide further insight.

For completeness, I can't talk to the programmer at all when using PCA10031, PCA10028, or PCA10040 (with v5 connectivity), but I believe that to be either a CDC-ACM driver issue or a bug in `pc-ble-driver`.  In the steps above I'm using a Waveshare BLE400 which uses a more reliable serial driver on the Linux side.

pabigot, 2018-05-25:
Looks like a race condition.  What I'm seeing with a locally built and instrumented connectivity app is that a BLE event comes in and ser_conn_ble_event_encoder() allocates the tx buffer for it, successfully queues it for send, and pauses the scheduler.

Normally the transmission completes and the scheduler is resumed() before the next command packet comes in.  For unknown reasons in some situations the transmission never occurs, i.e. there's no ser_hal_transport phy_events_handler() invocation for it.  I've seen up to 100 ms before the next event.

In the observed case the next command packet comes in and ser_conn_command_process() spins waiting for the tx buffer to become available, which never happens and so all main loop activity is blocked.

I now suspect the `3` is from a timeout in `pc-ble-driver` due to the BLE response never being received.

Still no idea why this is occurring.  I do have a lot of beacons babbling in my environment.

pabigot, 2018-05-26:
I have the following evidence that this is a bug in nrfutil.  The image below comes from an instrumented version of the connectivity application.
[image]

For orientation:
* **19 TxAlloc** signal goes high when `m_tx_buffer` in `ser_hal_transport` is allocated and goes low when it's freed.
* **20 EvtTx** signal goes high when the TX buffer is allocated for `ser_conn_ble_event_encoder`, and low when it's freed.
* **21 CmdTx** signal goes high when the TX buffer is allocated for `ser_conn_command_process`, and low when it's freed.
* **22 RxActive** signal goes high when `phy_events_handler` in `ser_hal_transport` sets `m_rx_state` to `HAL_TRANSP_RX_STATE_RECEIVING`, and low when `m_rx_state` is set to `HAL_TRANSP_RX_STATE_RECEIVED`.
* **2 RxProcess** goes high when `ser_conn_rx_process` calls `ser_conn_received_pkt_process` and low when the call returns.
* **3 TxActive** signal goes high when `ser_hal_transport_tx_pkt_send` in `ser_hal_transport` sets `m_tx_state` to `HAL_TRANSP_TX_STATE_TRANSMITTING`, and low when `phy_events_handler` sets it to `HAL_TRANSP_TX_STATE_TRANSMITTED`.

In most of the image event transmission and command transmission interleaves nicely.  Although a command reception (22) can overlap with an event transmission (20), the transmission completes before the command is processed (2).

At the right side this breaks:

* At 8.640 ms **20 EvtTx** goes high, indicating the transmit buffer is reserved for an event.  The HAL transport gets the `SER_PHY_HCI_SLIP_EVT_PKT_SENT` event and `hci_tx_fsm_event_process` enters `WAIT_FOR_ACK`.
* At 8.643 ms **22 RxActive** goes high, and then **2 RxProcess** gets to `ser_conn_command_process` where it spins waiting for the transmit buffer that's being used by the event.

For the transmit buffer to be released, a `SLIP/PKT_RECEIVED` event has to be processed.  This never happens: other instrumentation shows that `ser_phy_uart_evt_callback` processes no additional data after the command is received, so the fault seems to be on the host side.

My hypothesis is nrfutil is waiting in the loop for the response to the command that it sent, and there's no thread available to processes the pending SLIP queue.

If [Nordic case 207383](https://devzone.nordicsemi.com/f/nordic-q-a/34791/ble-connectivity-state-machine-error) is fixed the transmit ultimately fails and `SER_PHY_EVT_HW_ERROR` ultimately makes it back to the connectivity application, but by this point nrfutil has aborted with the traceback above.

If my analysis is correct, this is a bug in the host-side Python/Boost infrastructure, in that processing SLIP messages does not account for interleaved transactions.  I'm not sure why nobody else sees it or why it shows up so often in my environment; maybe it's due to recent changes and most folks aren't using the current code with nRF51 devices.  In any case, the best I can do without replacing the host-side implementation is detect that it's happened and restart the update process until the complete image gets transferred.

If my analysis is not correct I'd appreciate any clues that might help me find a better workaround.

bihanssen, 2018-05-29:
Thanks for reporting. We will look into this and report back.

josschne, 2018-08-15:
Looking for an update on this. Seeing the same error.
` Failed to ble_gattc_write. Error code: 3` 

OSX 10.13.6 (High Sierra)
nrfutil 3.5.1
Nordic SDK 15.0.0 on a 52840

bihanssen, 2018-08-16:
Hi,
the problem is likely related to congestion on the transport of the nRF5 connectivity. This would be more likely to happen on the nRF51, though I see you observe on nRF52 as well. There have been made some improvements to the state machine of the underlying pc-ble-driver, which have not made their way into the pc-ble-driver-py. 
I have prepared an alpha build of pc-ble-driver-py that it would be interesting to hear if makes any improvements to the issue you see.
[pc_ble_driver_py-0.11.4a0.tar.gz](https://github.com/NordicSemiconductor/pc-nrfutil/files/2293070/pc_ble_driver_py-0.11.4a0.tar.gz)
Procedure to install:
pip uninstall pc_ble_driver_py
pip install pc_ble_driver_py-0.11.4a0.tar.gz

Could you also give info on what kits (boards) you are using?

bihanssen, 2018-08-22:
@pabigot you mention that you use a Waveshare BLE400 board. Could you verify with the logic analyzer that the UART HW flow control line is being reacted to immediately by the onboard serial module, meaning it backs off from sending data when nRF5 chip raises RTS/CTS?

pabigot, 2018-08-22:
I'm not set up to test right now, but I did extensive testing on UART flow control back in May.  My notes record that the BLE400 does deassert RTS when the second byte is received and not read from the RXB register, and in this situation a Linux serial session with crtscts enabled does cause Linux to stop attempting to transmit.  I think that provides a "yes it does" answer to your question.

I have no reason to believe hardware flow control problems are involved in what I reported, though see below for what others might see.  The data support a hypothesis that the PC side isn't prepared to handle SLIP packets from multiple streams.

Do you have the potential patches to address this issue in branches on the github repositories?  If so, could you point me to them?  If I dig into this again I'd like to work with something more informative than a pre-packaged release.

(BTW: in my UART testing I confirmed [this behavior](https://devzone.nordicsemi.com/f/nordic-q-a/5200/does-the-uart-disable-rts-when-it-sees-cts-disabled) as well as numerous problems with ttyACMx communication on PCA10028 devices including loss of flow control signals when using certain USB hubs and writes above 64 bytes without MSD disabled on the OB_SAM3U.  So certainly UART firmware updates over ttyACMx with nRF5 development hardware on Linux are problematic.)

bihanssen, 2018-08-23:
> I have no reason to believe hardware flow control problems are involved in what I reported

Ok, good.

> Do you have the potential patches to address this issue in branches on the github repositories?

The .tar.gz I provided is simply a build of the current tip of master of pc-ble-driver-py, which includes an updated reference to latest pc-ble-driver submodule. It has not been pushed to pip, that's why I provided the file, though you could build it from source as well if you have the toolchain set up. 
I can't say for sure that the updated pc-ble-driver and connectivity fw will affect the problem you are seeing, but it might be related.

If you have the opportunity you could provide the verbose log for the failing dfu procedure, that could give great context info in addition to the plots you already provided. You can enable verbose logging with `nrfutil -vvvv dfu ble (...)`

Thanks.

gorazdko, 2018-09-17:
> If you have the opportunity you could provide the verbose log for the failing dfu procedure

```
C:\Users\Gorazd\Desktop\projects-7\nRF5_SDK_15.0.0_a53641a\examples\ble_peripheral\ble_app_hrs\hex>nrfutil -vvv  dfu ble -f -ic NRF52 -pkg phrm_2_2.zip
2018-09-17 13:16:04,144 Shared library: C:\Users\Gorazd\AppData\Local\Temp\_MEI19~1\lib\win\x86_32\pc_ble_driver_shared_sd_api_v3.dll
2018-09-17 13:16:04,144 Swig module name: pc_ble_driver_sd_api_v3
No target selected. Default device name: DfuTarg is used.
Please select connectivity serial port:
        0 : COM4 - 682952530
        1 : COM6 - 682342304
Enter your choice: : 1
Board already flashed with connectivity firmware.
2018-09-17 13:16:13,125 Using connectivity board at serial port: COM6
2018-09-17 13:16:13,137 Sending Application image.
2018-09-17 13:16:17,278
BLE: ble_enable with local ATT MTU: 247
2018-09-17 13:16:17,293 BLE: Scanning for DfuTarg
2018-09-17 13:16:17,318 Received advertisement report, address: 0xD9CAA3DA7935, device_name: DfuTarg
2018-09-17 13:16:17,319 BLE: Found target advertiser, address: 0xD9CAA3DA7935, name: DfuTarg
2018-09-17 13:16:17,319 BLE: Connecting to 0xD9CAA3DA7935
2018-09-17 13:16:17,328 Received advertisement report, address: 0xD9CAA3DA7935, device_name:
2018-09-17 13:16:17,329 BLE: Found target advertiser, address: 0xD9CAA3DA7935, name:
2018-09-17 13:16:17,329 BLE: Connecting to 0xD9CAA3DA7935
2018-09-17 13:16:17,338 Received advertisement report, address: 0xF88C416E7CA6, device_name: FF-F88
2018-09-17 13:16:17,345 Received advertisement report, address: 0xF88C416E7CA6, device_name:
2018-09-17 13:16:17,354 Received advertisement report, address: 0x3EA78B5D2F31, device_name:
2018-09-17 13:16:17,361 BLE: Connected to [217L, 202L, 163L, 218L, 121L, 53L]
2018-09-17 13:16:18,368 Successfully Connected
2018-09-17 13:16:18,371 BLE: Service Discovery
2018-09-17 13:16:18,401 New service uuid: 0x1800, start handle: 1, end handle: 9
2018-09-17 13:16:18,404 New service uuid: 0x1801, start handle: 10, end handle: 10
2018-09-17 13:16:18,405 New service uuid: 0xFE59, start handle: 11, end handle: 65535
2018-09-17 13:16:18,447 New characteristic uuid: 0x2A00, declaration handle: 2, value handle: 3
2018-09-17 13:16:18,450 New characteristic uuid: 0x2A01, declaration handle: 4, value handle: 5
2018-09-17 13:16:18,451 New characteristic uuid: 0x2A04, declaration handle: 6, value handle: 7
2018-09-17 13:16:18,489 New characteristic uuid: 0x2AA6, declaration handle: 8, value handle: 9
2018-09-17 13:16:18,565 New descriptor uuid: 0x2A00, handle: 3
2018-09-17 13:16:18,595 New descriptor uuid: 0x2A01, handle: 5
2018-09-17 13:16:18,641 New descriptor uuid: 0x2A04, handle: 7
2018-09-17 13:16:18,684 New descriptor uuid: 0x2AA6, handle: 9
2018-09-17 13:16:18,849 New characteristic uuid: 0x02, declaration handle: 12, value handle: 13
2018-09-17 13:16:18,894 New characteristic uuid: 0x01, declaration handle: 14, value handle: 15
2018-09-17 13:16:18,983 New descriptor uuid: 0x02, handle: 13
2018-09-17 13:16:19,029 New descriptor uuid: 0x01, handle: 15
2018-09-17 13:16:19,059 New descriptor uuid: 0x2902 (Standard.cccd), handle: 16
2018-09-17 13:16:19,101 BLE: Enabling longer ATT MTUs
2018-09-17 13:16:19,101 Sending GATTC MTU exchange request: 247
2018-09-17 13:16:19,134 GATTC: ATT MTU: 247
2018-09-17 13:16:19,134 ATT MTU exchanged: conn_handle=0 att_mtu=247
2018-09-17 13:16:19,144 BLE: Enabling Notifications
2018-09-17 13:16:19,220 BLE: Set Packet Receipt Notification 0
2018-09-17 13:16:19,265 Sending init packet...
2018-09-17 13:16:19,272 BLE: Selecting Object: type:1
2018-09-17 13:16:19,313 BLE: Object selected: max_size:256 offset:0 crc:0
2018-09-17 13:16:19,358 BLE: Streaming Data: len:135 offset:0 crc:0x00000000
2018-09-17 13:16:19,634 Sending firmware file...
2018-09-17 13:16:19,647 BLE: Selecting Object: type:2
2018-09-17 13:16:19,773 BLE: Object selected: max_size:4096 offset:0 crc:0
2018-09-17 13:16:19,821 BLE: Streaming Data: len:4096 offset:0 crc:0x00000000
Traceback (most recent call last):
  File "nordicsemi\__main__.py", line 987, in <module>
  File "site-packages\click\core.py", line 722, in __call__
  File "site-packages\click\core.py", line 697, in main
  File "site-packages\click\core.py", line 1066, in invoke
  File "site-packages\click\core.py", line 1066, in invoke
  File "site-packages\click\core.py", line 895, in invoke
  File "site-packages\click\core.py", line 535, in invoke
  File "nordicsemi\__main__.py", line 840, in ble
  File "nordicsemi\dfu\dfu.py", line 129, in dfu_send_images
  File "nordicsemi\dfu\dfu.py", line 102, in _dfu_send_image
  File "nordicsemi\dfu\dfu_transport_ble.py", line 549, in send_firmware
  File "nordicsemi\dfu\dfu_transport_ble.py", line 619, in __stream_data
  File "nordicsemi\dfu\dfu_transport_ble.py", line 332, in write_data_point
  File "site-packages\pc_ble_driver_py\ble_adapter.py", line 366, in write_cmd
pc_ble_driver_py.exceptions.NordicSemiException: Failed to ble_gattc_write. Error code: 3
Failed to execute script __main__
```

bihanssen, 2018-09-27:
@gorazdko, is your setup related to the setup that pabigot has reported? I see you run on Windows, and it seems like you use nRF52 devkit?

gorazdko, 2018-09-27:
> @gorazdko, is your setup related to the setup that pabigot has reported? I see you run on Windows, and it seems like you use nRF52 devkit?

Yes im using two nrf52832 dev boards, Windows 10, SDK15 and nrfutil 3.5.1.

Updating over nrf connect on my android phone works just fine though.

sijohans, 2018-11-05:
I have this issue on an ubuntu 16.04 machine and my osx device as well.

mikebwilliams, 2018-11-30:
I'm also having this issue on 18.04 using nrfutil 4.0.0.

bihanssen, 2018-12-03:
@gorazdko, @sijohans, @mikebwilliams, thanks for reporting.

It might be that either the DFU target or connectivity firmware is not able to handle the amount of data, in which case it would be interesting to see if it helps to reduce the throughput. 

We will need to look into this, though in the meantime it would be nice if someone could try to modify one or both of the following parameters in `Python27\Lib\site-packages\nordicsemi\dfu\dfu_transport_ble.py` (Windows path):
reduce `LOCAL_ATT_MTU` on line 87
increase `min_conn_interval_ms` on line 368

hecanbo, 2018-12-25:
I'm also having this issue on 18.04 using

AutopsyMania, 2019-01-08:
> @gorazdko, @sijohans, @mikebwilliams, thanks for reporting.
>
> It might be that either the DFU target or connectivity firmware is not able to handle the amount of data, in which case it would be [...]

reduce LOCAL_ATT_MTU to 200, increase min_conn_interval_ms to 15 does not help.
Ubuntu 18.0.4 nrfutil 4.0.0

bihanssen, 2019-01-08:
@AutopsyMania thanks for testing. ATT MTU of 200 is still large, 23 would be a better value to test with as it is the default size.

AlejandroHCruz, 2019-02-01:
I can confirm this also happens on the latest MacOS. Setting the MTU to 23 fixed the issue.

JavierGarJim, 2019-02-20:
Any update regarding this bug? It still happens to me working in macOS with the current master branch of nrfutil. It does not matter which MTU size I set.

bobatsar, 2019-02-21:
For me using the new -mtu argument with 23 fixes the problem on linux too, but the transfer is really slow.

alamysm, 2019-07-15:
I had nrfutil BLE DFU working on macOS without issue on an older nRF52 DK, but then when I got to the office and tried to use a brand new dev board, I started having this issue. Setting MTU to 23 fixed the issue, but as others mentioned, updates were significantly slower. But then I remembered about the MSD issue with jlink on OSX (https://wiki.segger.com/J-Link_OB), and after I ran the MSDDisable command in jlinkexe and cycled power, my new dev board appears to be back to working even with the MTU set to 247.

mrquincle, 2019-07-29:
I can confirm @alamysm's finding. 

JLinkExe
> MSDDisable

Now without changes to MTU, it runs flawlessly. Thanks so much!

- kernel: 4.15.0-55-generic
- os: Ubuntu 18.04.2 LTS
- SEGGER J-Link Commander V6.48 (Compiled Jul 26 2019 17:51:05)

bihanssen, 2019-10-18:
Closing as fixed. Please report if there are issues found in the latest version (nrfutil v6).

closed 2019-10-18 by bihanssen