NordicSemiconductor/pc-nrfutil #151: Error 13 during nrfutil dfu ble

MariusGailius, 2018-06-18:
I am trying to run ble dfu from ubuntu command line and it **always** reports error 13.

Setup
- Fresh install of Ubuntu 16.04 64-bit
- PCA10040 v1.1.1 dev kit as connectivity board
- nrfutil 3.5.1 installed with `pip install nrfutil`
- User added to `dialout` with `sudo usermod -a -G dialout margai`
- Modemmanager service disabled 

nrfutil is invoked with 
`nrfutil dfu ble -ic NRF52 -p /dev/ttyACM0 -pkg Rcn214dev-secure.zip -f`

Error:

> No target selected. Default device name: DfuTarg is used.
Flashing connectivity firmware...
Connectivity firmware flashed.
  [------------------------------------]    0%
Traceback (most recent call last):
  File "/home/margai/.local/bin/nrfutil", line 11, in <module>
    sys.exit(cli())
  File "/home/margai/.local/lib/python2.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/margai/.local/lib/python2.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/margai/.local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/margai/.local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/margai/.local/lib/python2.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/margai/.local/lib/python2.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/margai/.local/lib/python2.7/site-packages/nordicsemi/__main__.py", line 838, in ble
    dfu.dfu_send_images()
  File "/home/margai/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu.py", line 129, in dfu_send_images
    self._dfu_send_image(self.manifest.application)
  File "/home/margai/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu.py", line 90, in _dfu_send_image
    self.dfu_transport.open()
  File "/home/margai/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_ble.py", line 450, in open
    self.dfu_adapter.open()
  File "/home/margai/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_ble.py", line 106, in open
    self.adapter.driver.open()
  File "/home/margai/.local/lib/python2.7/site-packages/pc_ble_driver_py/ble_driver.py", line 126, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13

What I have tried:

- Connectivity hex files from [https://github.com/NordicSemiconductor/pc-ble-driver/tree/v2.2.1/hex](url). Same error is produced with both sd_api_v2 and sd_api_v3 hex files with 1M and 115200 baud rates. 
- Installing pc-ble-driver-py from source [https://github.com/NordicSemiconductor/pc-ble-driver](url). Still same error.

Other notes:

`nrfutil dfu ble` runs fine on windows machine with the same PCA10040 board. nrfutil version 3.5.1

I get same error when running `nrfutil dfu ble` on Raspberry Pi

I have never succeeded to perform dfu update on Linux machine

Advertising example from pc-ble-driver-py produces error 13 too. Invoked with `python advertising.py NRF52 /dev/ttyACM0`. Error:
>python advertising.py NRF52 /dev/ttyACM0
Serial port used: /dev/ttyACM0
Traceback (most recent call last):
  File "advertising.py", line 79, in <module>
    main(sys.argv[2])
  File "advertising.py", line 55, in main
    driver.open()
  File "/home/margai/.local/lib/python2.7/site-packages/pc_ble_driver_py/ble_driver.py", line 126, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13

Any ideas what to try next?

bobatsar, 2018-09-11:
I had the same issue and an update of the [segger tools](https://www.segger.com/downloads/jlink#J-LinkSoftwareAndDocumentationPack) fixed the problem.

bihanssen, 2018-10-17:
Closing this issue as it seems resolved. Please repost on devzone if there is still a problem

closed 2018-10-17 by bihanssen