NordicSemiconductor/pc-nrfutil #153: AttributeError: 'int' object has no attribute 'encode'

louisld, 2018-07-25:
When I try to create a dfu package for the ruuvitag with :
`$ nrfutil pkg generate --application _build/ruuvi_firmware.hex --hw-version 3 --sd-req 0x91 --key-file ~/dev/ruuvitag_fw/keys/ruuvi_open_private.pem --application-version 1 ruuvi_firmware_dfu.zip
` I get a python error.
```Traceback (most recent call last):

  File "/usr/local/bin/nrfutil", line 11, in <module>

    sys.exit(cli())

  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 722, in __call__

    return self.main(*args, **kwargs)

  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 697, in main

    rv = self.invoke(ctx)

  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1066, in invoke

    return _process_result(sub_ctx.command.invoke(sub_ctx))

  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1066, in invoke

    return _process_result(sub_ctx.command.invoke(sub_ctx))

  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 895, in invoke

    return ctx.invoke(self.callback, **ctx.params)

  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 535, in invoke

    return callback(*args, **kwargs)

  File "/usr/local/lib/python3.6/dist-packages/nordicsemi/__main__.py", line 617, in generate
    default_key = signer.load_key(key_file)

  File "/usr/local/lib/python3.6/dist-packages/nordicsemi/dfu/signing.py", line 92, in load_key
    sk_hex = "".join(c.encode('hex') for c in self.sk.to_string())

  File "/usr/local/lib/python3.6/dist-packages/nordicsemi/dfu/signing.py", line 92, in <genexpr>
    sk_hex = "".join(c.encode('hex') for c in self.sk.to_string())

AttributeError: 'int' object has no attribute 'encode'
```

closed 2018-07-25 by louisld

mounthill, 2018-08-13:
how did you solve it?

louisld, 2018-08-13:
I had a problem with the installation of some module with pip on debian. I thought it was using python 3 but it was using python 2. I had the both version installed. The issue was work apt package. I installed pip via python and it worked fine.

hecko, 2019-10-16:
Im hitting the same issue om MacOS Catalina with Python3 from brew and nrfutils installed as

`pip3 install nrfutils`

This is the result:

`Marcels-MBP-511:dfu marcel$ /Users/marcel/Library/Python/3.7/bin/nrfutil pkg generate --hw-version 52 --sd-req 0xB8 --application-version 1 --application ../../FW/norman_5_firmware/pca10056e/s112/ses/Output/Release/Exe/ble_app_uart_pca10056e_s112.hex --key-file key.pem norman5_dfu_v1.zip
Traceback (most recent call last):
  File "/Users/marcel/Library/Python/3.7/bin/nrfutil", line 10, in <module>
    sys.exit(cli())
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/nordicsemi/__main__.py", line 815, in generate
    default_key = signer.load_key(key_file)
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/nordicsemi/dfu/signing.py", line 92, in load_key
    sk_hex = "".join(c.encode('hex') for c in self.sk.to_string())
  File "/Users/marcel/Library/Python/3.7/lib/python/site-packages/nordicsemi/dfu/signing.py", line 92, in <genexpr>
    sk_hex = "".join(c.encode('hex') for c in self.sk.to_string())
AttributeError: 'int' object has no attribute 'encode'`

Any thoughts?