NordicSemiconductor/pc-nrfutil #158: Python 3 Support

cooperlees, 2018-09-06:
Hi,

Has any effort been made or planned for Python 3 support? I'm also willing to help make this happen.
- Are there any hard blockers? If so, what are they?
- Due to your strong Python 2 user base, naturally I would like to do it to support >=2.7 + Python 3

Reason:
Looking to use nrfutil on a Yocto Sumo system that we're planning to make Python 3 only asap.

Cheers,
Cooper

cooperlees, 2018-09-12:
Ok - Thanks for the pointers. I'll see if I can start to get some PRs happening.
- Totally can support Python 2 and 3 for now
- As the other thread says we can aim to have a universal wheel that supports Py2 and 3, or we have separate wheels. I'll be more informed of which makes more sense when I look at the code.

I'll start work on pc-ble-driver-py first. I should of searched closed as well before opening this - You happy to keep this open or work from #40 ?

bihanssen, 2018-09-13:
Hi,
you can keep this open, that's no problem.
Thanks for the initiative, call out if any questions.

BI

shalak, 2018-09-20:
Hi!

pc-ble-driver-py seems to support python3 since [PR#50](https://github.com/NordicSemiconductor/pc-ble-driver-py/pull/50) from April
pyspinel support python3 since [PR#45](https://github.com/openthread/pyspinel/pull/45) from couple of days ago

Making `nrfutil` work with py3 will be well received by everyone who utilizes it in their frameworks.

bihanssen, 2018-10-17:
@shalak pc-ble-driver-py does not support python3, PR#50 did not add all necessary changes.
We will start work with adding py3 support work after releasing next version of pc-ble-driver.

bihanssen, 2019-10-18:
Python 3 support has been added as of nrfutil v6.

closed 2019-10-18 by bihanssen

Terstegge, 2021-04-09:
I just installed nrfutil using
pip3 install nrfutil
and get an error after running the tool:

Traceback (most recent call last):
  File "/home/andreas/.local/bin/nrfutil", line 5, in <module>
    from nordicsemi.__main__ import cli
  File "/home/andreas/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 49, in <module>
    from nordicsemi.dfu.bl_dfu_sett import BLDFUSettings
  File "/home/andreas/.local/lib/python3.8/site-packages/nordicsemi/dfu/bl_dfu_sett.py", line 51, in <module>
    from nordicsemi.dfu.package import Package
  File "/home/andreas/.local/lib/python3.8/site-packages/nordicsemi/dfu/package.py", line 53, in <module>
    from nordicsemi.dfu.init_packet_pb import InitPacketPB, DFUType, CommandTypes, ValidationTypes, SigningTypes, HashTypes
  File "/home/andreas/.local/lib/python3.8/site-packages/nordicsemi/dfu/init_packet_pb.py", line 38, in <module>
    from . import dfu_cc_pb2 as pb
  File "/home/andreas/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_cc_pb2.py", line 18, in <module>
    DESCRIPTOR = _descriptor.FileDescriptor(
TypeError: __init__() got an unexpected keyword argument 'serialized_options'

So what can I do here??