NordicSemiconductor/pc-nrfutil #160: Thread DFU using ethernet

MaryFrantz, 2018-09-11:
I integrated the Thread Secure DFU Example into my nRF52840 project and am able to download firmware using nrfutil.  From the documentation: "For example, the DFU server could be a cloud service, an nRF52840 Development Kit in conjunction with nrfutil, or a mobile phone running an application."

I don't want to ship my customer a Development Kit in order to perform an update in the field.  So, I'm wanting a server that would run at my customer's site on a host computer.  The same computer that would be communicating with the Thread nodes via the Border Router.  Or, a server that would run on the Border Router (Raspberry Pi + NCP).

Are there any plans to modify nrfutil to provide a DFU server over ethernet? Or on Raspberry Pi?

See also my ticket on the Nordic Dev Zone:
 https://devzone.nordicsemi.com/f/nordic-q-a/36891/nrfutil-for-border-router

I was referred to this site so that I might modify the nrfutil myself.  I attempted replacing the NCPTransport with a class that uses sockets, but it didn't work.  I have no experience with Python, so am having difficulty with it.

Mary

LuDuda, 2018-09-14:
Hello @MaryFrantz, as you already pointed out, adding a new transport next to NCPTransport should be viable. Unfortunately we don't have currently plans to add this support.

bihanssen, 2018-09-17:
Closing as answered. Please reopen if needed.

closed 2018-09-17 by bihanssen