NordicSemiconductor/pc-nrfutil #166: COAP DFU: Request time out error on thread

arn-soft, 2018-09-28:
Issue: Unable to Securely DFU over thread network.
Details: nrfutil is not sending response properly. which leads to "COAP DFU : Request time out" after device send trigger request. 
Discussion: [Link](https://devzone.nordicsemi.com/f/nordic-q-a/33803/thread-secure-dfu-example-not-working-on-nrf5-sdk-for-thread-and-zigbee-v1-0-0/150801#150801)
**Setup:**
Board: nrf52840 DK
nrfutil version: tried with both 3.5.1 as well as 4.0.0
nrf sdk for zigbee and thread: tried both 1.0.0 as well as 2.0.0
os: ubuntu 16.04

arn-soft, 2018-09-28:
Logs from DFU client and DFU server has been attahed.
[log_dfu_server.txt](https://github.com/NordicSemiconductor/pc-nrfutil/files/2427445/log_dfu_server.txt)
[log_dfu_client.txt](https://github.com/NordicSemiconductor/pc-nrfutil/files/2427466/log_dfu_client.txt)

arn-soft, 2018-10-04:
I captured the coap packets on thread network. It is strange that i haven't got any coap packet from dfu server(nrfutil). although i am being able to receive the coap packet of coap client (nrf52840-DK).

kpyush, 2018-10-15:
Any update on this issue?

arn-soft, 2018-10-15:
Any update on this issue?

e-rk, 2018-10-16:
Hi arn-soft,
There is now a PR which may solve your issue (https://github.com/NordicSemiconductor/pc-nrfutil/pull/169).
Can you please try it out and let us know if it works? It will be a valuable feedback.
Thank you.

bihanssen, 2018-10-16:
@arn-soft I will check up and get back to you

bihanssen, 2018-10-17:
@arn-soft, the PR from @e-rk is the current proposed fix. Does it seem to fix the problem you were seeing?

arn-soft, 2018-10-18:
@bihanssen @e-rk I have checked the pr(#169 ) for both unicast as well as multicast mode. Following are my observations.
1. `Unicast mode`:
          It is still not working in unicast mode and throwing **<error> nrf_dfu_validation: Failed to decode init packet** error. I am attaching dfu server as well client log.
[image]
[image]

2. `Multicast mode`
          It is working with a very slow rate of around 1 bit/sec. if i increase the rate (-r ) to even 4, then it fails to download. In multicast, If device misses any blocks, it can ask dfu server to resend the missed block. I don't think that is working properly either. Logs attached.
[image]
[image]

e-rk, 2018-10-18:
Hi,
Can you please verify that pyspinel 1.0.0a2 installed in your python2 site-packages? Also what is the command you are using to run nrfutil for both unicast and multicast?
The issue with device requesting resend in multicast DFU is strange. Can you elaborate on that?

arn-soft, 2018-11-22:
Hi @e-rk sorry for the delayed response. I followed every steps you have mentioned. I am still getting the same issue.

1.  Currently i am using two boards (one as a DFU server and one as a dfu client).
2.  I have tried with both sed (with rx enable on idle ) and ftds.
3.  DK version 1.0.0, dfu client (example in nRF5_SDK_for_Thread_and_Zigbee_2.0.0_29775ac).

I have been stuck on this issue for very long time. Please help me out. i will be more than happy to give you remote access of the whole set up.

pszkotak, 2018-11-26:
Hello @arn-soft ,
Thank you for your patience and understanding.

I checked unicast and multicast logs you provided and have two observations.
1. It is odd that you observe `nrf_dfu_validation: Failed to decode init packet`. Can you confirm that you run all tests on a fresh repository (git clean -xdf could help here, but beware, many files will be deleted)? If the `dfu_public_key.c` has been generated and copied before compiling the bootloader and the client it should work well. 

2. Your transfer is indeed quite slow. It is rather in blocks per second, not bits, but still it should be higher in proper conditions. I am wondering if you could experiment with different Thread channels. Maybe there is heavy 2.4GHz (e.g. older versions of Wi-Fi) in your area. Channel expected to work the best with a standard Wi-Fi is usually the 15. Channels 20 and up could be a good shot too.
Could you also try with different development kits to make sure that it is not the hardware issue?

Instruction posted by by @e-rk works well in our environment. This is why we struggle so much to isolate the issue you observed. Trying different channel, developemnt kits, carefully redoing setup with the cleaned repository (or even fresh SDK download) would be my starting point.

Feel free to share any thoughts and observations you have, any hints may push us to the right track.

Kind regards,
Piotr

bihanssen, 2019-10-18:
Closing due to inactivity. Please reopen if needed.

closed 2019-10-18 by bihanssen