NordicSemiconductor/pc-nrfutil #168: DFU commands over a USB serial connection are different between Windows and other OS

sekigon-gonnoc, 2018-10-09:
Althogh both of nrfutil are same version(4.0.0), the dfu command over a USB serial are different.
It is `nrfutil.exe dfu usb_serial` for Windows and is `nrfutil dfu usb-serial` for other OS.

bihanssen, 2018-10-17:
Hi, if you use the precompiled nrfutil.exe then you are correct, you would have to call the nrfutil.exe. However if you use pip install nrfutil then it will behave the same for all systems, including Windows.

sekigon-gonnoc, 2018-10-17:
I confirmed nrfutil installed by pip has the option `usb-serial` on Windows.
The option `usb_serial` should be modified to `usb-serial` in the next release of precompiled nrfutil.exe.

bihanssen, 2018-10-17:
I don't know why the .exe variant uses underscore and not the other variants, they are built from the same source. I will have to look into it.

bihanssen, 2018-10-17:
@sekigon-gonnoc I can confirm your findings, partly. I have found that on Ubuntu linux it shows usb-serial (non-underscore), while on Windows (both .exe and pip variants) and mac it shows usb_serial (underscore). I don't know why they behave differently, it seems pretty random at this point.

PR #186 (2019-02-08): Ensure same command name on different operating systems
Command names with underscores have been observed to end up as hyphens on Windows and Mac, and underscores on Linux. Specifying the "name" property ensures correct naming.
Fixes #168.
closed 2019-02-08 by bihanssen