NordicSemiconductor/pc-nrfutil #170: Suppress giant warning?

jimparis, 2018-10-19:
The new warning added by f9cec1ed1 seems overly obnoxious.  I am targetting SDK 15.1 and therefore do want the backup page, but there appears to be no way to disable this huge 15 line warning in my build output every time.  Can it be toned down a bit?

bihanssen, 2018-10-19:
Hi, I see your point regarding the warning. We will consider how it could be modified. If you have a suggestion you could add it in a PR.
Thanks.

jimparis, 2018-10-19:
Sure, I added a PR, although really this warning should disappear entirely at some point.  Maybe nrfutil should be passed an option specifying which SDK version it should target, and then it can make these sorts of decisions automatically.

bihanssen, 2018-10-23:
Thanks, I added your suggestion about specifying SDK version to our backlog for evaluation.

PR #185 (2019-02-07): Tone down warning about overwriting nRF DFU settings backup page
Closes #171.
Closes #170.
closed 2019-02-07 by bihanssen