NordicSemiconductor/pc-nrfutil #172: how to do DFU over mesh for common modules without Segger ID ?

zzzh, 2018-11-02:
In DFU quick start guide, it's said: This guide assumes that you have at least 2 development kits or devices, where one is to be interfaced over the serial port, while the other receives the DFU from the first device over the mesh. To specify which device to use in which context, add the -s <serial-number> option for each call to the nrfjprog command, where <serial-number> is the Segger ID of your device. This will make nrfjprog execute its operations on the specified device only.

But you can't expect the modules for common use ( instead of develop use ) have Segger ID, right?

bihanssen, 2018-11-06:
@zzzh, it seems you are commenting on the Mesh DFU documentation. Please instead refer to [nRF5 SDK for Mesh](https://github.com/NordicSemiconductor/nRF5-SDK-for-Mesh#reporting-issues).

closed 2018-11-06 by bihanssen

Beul, 2019-01-28:
This question is not specific to Mesh-DFU and also applies to other cases. For example if I deploy for my clients 2 devices and I want to update the application of my device A (without serial/USB connector) from my device B (with USB) over BLE. 

In this case my device B does not have Segger ID ?