NordicSemiconductor/pc-nrfutil #181: nrfutil treats device address and name parameters as an OR instead of an AND

mikebwilliams, 2018-11-29:
https://github.com/NordicSemiconductor/pc-nrfutil/blob/a305f5a8ec7bc2deda91829cdf7ac18e7f48217b/nordicsemi/dfu/dfu_transport_ble.py#L367

When I specify both a device name of "" and an address, nrfutil mistakenly connects to the first device that has a matching empty name and doesn't check that the address matches.

Example command and output:
./nrfutil -v -v dfu ble --package ../out//test_dfu_package.zip -ic NRF52 --jlink_snr 682803006 --address FF:BC:ED:A2:AC:0E -p /dev/ttyACM0 -f -n "" 2>&1 | tee logfile | less
2018-11-29 14:01:26,106 Shared library: /tmp/_MEIuMDGvn/lib/linux/x86_64/libpc_ble_driver_shared_sd_api_v3.so
2018-11-29 14:01:26,106 Swig module name: pc_ble_driver_sd_api_v3
Board already flashed with connectivity firmware.
2018-11-29 14:01:28,605 Using connectivity board at serial port: /dev/ttyACM0
2018-11-29 14:01:28,614 Sending Application image.
2018-11-29 14:01:32,174
BLE: ble_enable with local ATT MTU: 247
2018-11-29 14:01:32,287 BLE: Scanning for
2018-11-29 14:01:32,326 Received advertisement report, address: 0x001FFF4EA95D, device_name: PR BT 58CF
2018-11-29 14:01:32,331 Received advertisement report, address: 0x001FFF4EA95D, device_name:
2018-11-29 14:01:32,331 BLE: Found target advertiser, address: 0x001FFF4EA95D, name:
2018-11-29 14:01:32,331 BLE: Connecting to 0x001FFF4EA95D
2018-11-29 14:01:36,171 BLE: Connected to [0L, 31L, 255L, 78L, 169L, 93L]

bihanssen, 2018-12-21:
The behavior is not a bug but is intended. It is not described anywhere afaik that when both parameters are specified they shall be treated with AND. They are instead treated as a broadening of the range of devices to accept.

That said, it could be argued that combining with AND could be more useful in most scenarios. Will consider a change in implementation.

bihanssen, 2019-10-18:
Closing as won't fix. Changing the behavior would be a breaking change.

closed 2019-10-18 by bihanssen