NordicSemiconductor/pc-nrfutil #183: pyinstaller on mac failes with ValueError: Unknown Mach-O header

blasagna, 2018-12-20:
I am trying to compile nrfutil into a standalone executable, following the instructions in the README.

This works on Linux, and for Windows there are standalone executable attached to the nrfutil releases, but for mac I get the following error when attempting to use pyinstaller to generate a self-contained executable. 

Is there a specific version of pyinstaller that must be used? I have tried the instructions for nrfutil v3.5.1 and v4.0.0 with the same results. 

There is an old github issue with the same issue here that was closed without any follow-up: https://github.com/NordicSemiconductor/pc-nrfutil/issues/118

I've created a post on [DevZone](https://devzone.nordicsemi.com/f/nordic-q-a/41679/nrfutil-errors-on-macos) but am logging this here as well in case the developers do not get support tickets through that site.
```
$ pyinstaller ./nrfutil.spec 
INFO: PyInstaller: 3.4
INFO: Python: 2.7.15
INFO: Platform: Darwin-17.5.0-x86_64-i386-64bit
INFO: UPX is not available.
INFO: Extending PYTHONPATH with paths ['/Users/user/pc-nrfutil']
INFO: checking Analysis
INFO: Building because inputs changed
INFO: Initializing module dependency graph...
INFO: Initializing module graph hooks...
INFO: running Analysis Analysis-00.toc
INFO: Caching module hooks...
INFO: Analyzing nordicsemi/__main__.py
INFO: Processing pre-find module path hook   distutils
INFO: Processing pre-safe import module hook   _xmlplus
INFO: Loading module hooks...
INFO: Loading module hook "hook-distutils.py"...
INFO: Loading module hook "hook-sysconfig.py"...
INFO: Loading module hook "hook-xml.py"...
INFO: Loading module hook "hook-httplib.py"...
INFO: Loading module hook "hook-encodings.py"...
INFO: Loading module hook "hook-pkg_resources.py"...
INFO: Processing pre-safe import module hook   win32com
INFO: checking Tree
INFO: Building Tree because Tree-00.toc is non existent
INFO: Building Tree Tree-00.toc
INFO: Looking for ctypes DLLs
INFO: Analyzing run-time hooks ...
INFO: Including run-time hook 'pyi_rth_pkgres.py'
INFO: Including run-time hook 'pyi_rth_multiprocessing.py'
INFO: Looking for dynamic libraries
INFO: Looking for eggs
INFO: Using Python library /Users/api_team/miniconda3/envs/nrfutil/lib/libpython2.7.dylib
INFO: Warnings written to /Users/api_team/pc-nrfutil/build/nrfutil/warn-nrfutil.txt
INFO: Graph cross-reference written to /Users/api_team/pc-nrfutil/build/nrfutil/xref-nrfutil.html
INFO: Appending 'binaries' from .spec
INFO: checking PYZ
INFO: Building because toc changed
INFO: Building PYZ (ZlibArchive) /Users/api_team/pc-nrfutil/build/nrfutil/PYZ-00.pyz
INFO: Building PYZ (ZlibArchive) /Users/api_team/pc-nrfutil/build/nrfutil/PYZ-00.pyz completed successfully.
INFO: checking PKG
INFO: Building PKG because PKG-00.toc is non existent
INFO: Building PKG (CArchive) PKG-00.pkg
Traceback (most recent call last):
  File "/Users/api_team/miniconda3/envs/nrfutil/bin/pyinstaller", line 11, in <module>
    sys.exit(run())
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/__main__.py", line 111, in run
    run_build(pyi_config, spec_file, **vars(args))
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/__main__.py", line 63, in run_build
    PyInstaller.building.build_main.main(pyi_config, spec_file, **kwargs)
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/build_main.py", line 838, in main
    build(specfile, kw.get('distpath'), kw.get('workpath'), kw.get('clean_build'))
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/build_main.py", line 784, in build
    exec(text, spec_namespace)
  File "<string>", line 37, in <module>
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/api.py", line 424, in __init__
    strip_binaries=self.strip, upx_binaries=self.upx,
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/api.py", line 196, in __init__
    self.__postinit__()
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/datastruct.py", line 158, in __postinit__
    self.assemble()
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/api.py", line 259, in assemble
    dist_nm=inm)
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/building/utils.py", line 228, in checkCache
    dylib.mac_set_relative_dylib_deps(cachedfile, dist_nm)
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/depend/dylib.py", line 308, in mac_set_relative_dylib_deps
    dll = MachO(libname)
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/macholib/MachO.py", line 95, in __init__
    self.load(fp)
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/macholib/MachO.py", line 110, in load
    self.load_header(fh, 0, size)
  File "/Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/macholib/MachO.py", line 147, in load_header
    header, fh))
ValueError: Unknown Mach-O header: 0x230d0a23 in <open file '/Users/api_team/Library/Application Support/pyinstaller/bincache00_py27_64bit/lib/linux/__init__.py', mode 'rb' at 0x10e642150>
```

bihanssen, 2019-02-01:
Update: the .spec file is written in a way that is not compatible with being run on mac (with Mach-O library) and will need to be updated. Specifically it is the use of the _binaries_ argument that needs to be modified to avoid passing in irrelevant files.

closed 2019-02-13 by mathiasap

j9ac9k, 2019-10-24:
I just created a PR for pyinstaller with a hook that will grab `libportaudio.dylib` and place it in the correct spot; you can follow the progress here: https://github.com/pyinstaller/pyinstaller/pull/4498

mamur9, 2020-08-07:
@bihanssen Can you please give some example how to modify .spec file. I am having same issue while trying to add Tesseract-OCR binary.

bihanssen, 2020-08-07:
@mamur9, not sure how I can assist. We resolved the Mach-O header errors by switching to adding files with DATAS instead of BINARIES because else on MacOS it fails when being fed non-binary files. My best advice is likely to ask you to look at our current .spec file because that one works for us.

PR #196 (2019-02-13): Libusb and Pyinstaller
* Libusb1 binaries bundled with pyinstaller
* Pyinstaller works on MacOS
* Resolves correct paths for libusb1 binaries on Windows and MacOS. 

Closes #183 
Closes #139