NordicSemiconductor/pc-nrfutil #203: Thread DFU with different thread networking parameters

ozanoner, 2019-02-19:
There is no way to join to a thread network with different parameters:
Support needed for the following options:
- networkname
- extpanid
- masterkey

https://github.com/NordicSemiconductor/pc-nrfutil/blob/6267169e65c2d2645074c2bc758f914b7a26b789/nordicsemi/thread/tncp.py#L95

pszkotak, 2019-02-26:
Hello, @ozanoner .
Thank you for your interest in the Thread DFU.

The masterkey option has already been added by @andihas in PR https://github.com/NordicSemiconductor/pc-nrfutil/pull/182.

Would you like to make a followup PR with remaining parameters?

Kind regards,
Piotr

ozanoner, 2019-03-05:
Thank you @pszkotak and @andihas ! Yes, I think I can add remaining.
Best wishes

pszkotak, 2019-03-06:
Thank you @ozanoner  :+1: 
We will wait for your PR then.

Best regards,
Piotr

kenr, 2023-01-17:
We do not have capacity to maintain and release new versions of this utility.
A new utility with the same name and feature set is made. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr