NordicSemiconductor/pc-nrfutil #207: Merge mesh_dfu into master branch

igor-gorjanc, 2019-03-05:
Currently there are separate versions of nrfutil for clasic Bluetooth and mesh. The master branch supports almost all protocols (Bluetooth, Zigbee, ANT...) EXCEPT Mesh.  Using two separate versions of the same utility with the same name and almost the same syntax is very disturbing and has also resulted in a lot of support requests on the Nordic forum.
It would be really good to merge mesh_dfu into master branch.

bihanssen, 2019-03-25:
@igor-gorjanc thanks for reporting. You are right that it would probably be good to merge the branch. We will look into the issue and the priority.

igor-gorjanc, 2019-10-16:
Is there anything new regarding this? When can we expect a merge?

bihanssen, 2019-10-21:
@igor-gorjanc, I have spoken with the mesh developers, and the message is that there will likely be no merge in the foreseeable future due to the amount of work required to port the mesh branch to master. However, if you have the opportunity to do a PR on it they will take it up for review and approval.
Closing this issue.

closed 2019-10-21 by bihanssen