NordicSemiconductor/pc-nrfutil #209: Not able to install nrfutil in Mac OS X Darwin

iqramali, 2019-03-05:
I doesn't detect the OS X, can someone help me how to fix this issue?

```
sudo -H pip install --ignore-installed six nrfutil
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting six
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting nrfutil
  Using cached https://files.pythonhosted.org/packages/dd/2a/7758a00fa59419f38f8e902cbb9ba4dd77c0fb0b0ad831ea3819d216c9c6/nrfutil-5.1.0.tar.gz
    Complete output from command python setup.py egg_info:
    /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'zipfile'
      warnings.warn(msg)
    /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'console'
      warnings.warn(msg)
    error in nrfutil setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected ',' or end-of-list in antlib ~= 1.0b ; sys_platform == 'win32' at  ; sys_platform == 'win32'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in `/private/tmp/pip-install-kgzadh/nrfutil/
```

mathiasap, 2019-03-13:
@iqramali This seems to be a compatibility issue with older versions of distutils. Can you try to upgrade setuptools and see if it solves the problem?

bihanssen, 2019-03-25:
Closing as answered.

closed 2019-03-25 by bihanssen