NordicSemiconductor/pc-nrfutil #210: When running on W10 I get:  The 'antlib~=1.0b' distribution was not found and is required by nrfutil

mmonson, 2019-03-05:
I just pulled commit 1d20b11  and installed it on Windows 10 Pro Version 1809 
When I run nrfutil --help I get this error: 

File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 784, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'antlib~=1.0b' distribution was not found and is required by nrfutil

cchengVA, 2019-03-05:
FYI - make sure you're running a 32bit version of Python... we hit the same issue just now and that is what's fixing it.

bihanssen, 2019-03-07:
@mmonson thanks for reporting. Are you running 32 or 64 bit version of Python? 
For reproduction, can you describe how you install nrfutil after pulling the commit?

PR #218 (2019-03-29): Update requirements.txt
The pip requirements file syntax is not able to specify the requirement of Windows 32 bit Python, which is the only supported variant of antlib currently. Removing the entry from the requirements file as a workaround, and instead relying on manual installation of antlib.

Closes #210 and closes #213.
closed 2019-03-29 by bihanssen