NordicSemiconductor/pc-nrfutil #211: 5.1.0 broken. Related to libusb.

hallmatthewp, 2019-03-06:
After upgrading to nrfutil 5.1.0 (previously using 5.0.0 with great success), I encounter this when trying to run nrfutil:

`python2.7 ~/.local/bin/nrfutil settings generate --family NRF52840 --application _build/nrf52840_xxaa.hex --application-version 1 --bootloader-version 1 --bl-settings-version 1 bootloader_settings_bypass.hex
Traceback (most recent call last):
  File "/home/circleci/.local/bin/nrfutil", line 6, in <module>
    from nordicsemi.__main__ import cli
  File "/home/circleci/.local/lib/python2.7/site-packages/nordicsemi/__main__.py", line 53, in <module>
    from nordicsemi.dfu.dfu_transport_serial import DfuTransportSerial
  File "/home/circleci/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 53, in <module>
    from nordicsemi.dfu.dfu_trigger import DFUTrigger
  File "/home/circleci/.local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_trigger.py", line 60, in <module>
    import usb1
  File "/home/circleci/.local/lib/python2.7/site-packages/usb1/__init__.py", line 61, in <module>
    from . import libusb1
  File "/home/circleci/.local/lib/python2.7/site-packages/usb1/libusb1.py", line 199, in <module>
    libusb = _loadLibrary()
  File "/home/circleci/.local/lib/python2.7/site-packages/usb1/libusb1.py", line 173, in _loadLibrary
    return dll_loader('libusb-1.0' + suffix, **loader_kw)
  File "/usr/local/lib/python2.7/ctypes/__init__.py", line 366, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libusb-1.0.so: cannot open shared object file: No such file or directory`

bihanssen, 2019-03-07:
@hallmatthewp, thanks for reporting. Can you describe how you installed nrfutil 5.1.0, specifically did you use pip with or wo --user and with or wo sudo?

hallmatthewp, 2019-03-08:
I performed `pip install nrfutil --user`

To fix the issue, I now use `pip install nrfutil==5.0.0 --user`

mathiasap, 2019-03-13:
@hallmatthewp: Which Linux distribution are you running? I am assuming that you are using a distro that does not ship libusb1 binaries, which explains why you are getting a libusb-1.0.so loading error. Libusb1 is only required to trigger the bootloader from an application, and should not be strictly needed otherwise. Thanks for letting us know.

closed 2019-03-13 by mathiasap

hallmatthewp, 2019-03-19:
I'm running a docker container of Ubuntu 14.04. I'm curious what changed between 5.0.0 and 5.1.0, that suddenly requires libusb1 to be installed beforehand.

bihanssen, 2019-03-20:
@hallmatthewp the dependency to libusb was introduced in PR #190. Libusb is here utilized for usb enumeration and communication wrt. DFU.

PR #215 (2019-03-13): Missing libusb1 binaries gives a warning instead of failing.
Fixes #211