NordicSemiconductor/pc-nrfutil #212: Executable from pyinstaller fails on mac osx

blasagna, 2019-03-07:
On release v5.1.0, the pyinstaller build succeeds, but I get an error whenever I try to run the executable it produces.  This looks like it is likely related to this open [pyinstaller issue](https://github.com/pyinstaller/pyinstaller/issues/3198). Is there a known working version of pyinstaller the Nordic team is working with on mac?

I am attempting to package nrfutil in a virtual environment with Python 2.7.15 on mac OS X 10.13.4.

executable error:
```
$ ./dist/nrfutil --help
Traceback (most recent call last):
 File "site-packages/PyInstaller/loader/rthooks/pyi_rth_pkgres.py", line 11, in <module>
 File "/private/var/folders/d9/pt6kgkc939n4_4v60shh4c6w0000gn/T/pip-install-TcNthk/pyinstaller/PyInstaller/loader/pyimod03_importers.py", line 395, in load_module
 File "site-packages/pkg_resources/__init__.py", line 959, in <module>
 File "site-packages/pkg_resources/__init__.py", line 963, in Environment
 File "site-packages/pkg_resources/__init__.py", line 190, in get_supported_platform
 File "site-packages/pkg_resources/__init__.py", line 395, in get_build_platform
 File "sysconfig.py", line 618, in get_platform
 File "sysconfig.py", line 482, in get_config_vars
 File "sysconfig.py", line 365, in _init_posix
ImportError: No module named _sysconfigdata
[52012] Failed to execute script pyi_rth_pkgres
```

pyinstaller output:
```
$ pyinstaller --clean nrfutil.spec
29 INFO: PyInstaller: 3.4
30 INFO: Python: 2.7.15
35 INFO: Platform: Darwin-17.5.0-x86_64-i386-64bit
41 INFO: UPX is not available.
41 INFO: Removing temporary files and cleaning cache in /Users/api_team/Library/Application Support/pyinstaller
54 INFO: Extending PYTHONPATH with paths
['/Users/api_team/pc-nrfutil']
54 INFO: checking Analysis
54 INFO: Building Analysis because Analysis-00.toc is non existent
54 INFO: Initializing module dependency graph...
56 INFO: Initializing module graph hooks...
111 INFO: running Analysis Analysis-00.toc
116 INFO: Caching module hooks...
120 INFO: Analyzing nordicsemi/__main__.py
2540 INFO: Processing pre-find module path hook   distutils
3404 INFO: Processing pre-safe import module hook   _xmlplus
3949 INFO: Processing pre-safe import module hook   six.moves
7469 INFO: Loading module hooks...
7469 INFO: Loading module hook "hook-distutils.py"...
7470 INFO: Loading module hook "hook-sysconfig.py"...
7471 INFO: Loading module hook "hook-xml.py"...
7539 INFO: Loading module hook "hook-httplib.py"...
7540 INFO: Loading module hook "hook-encodings.py"...
8063 INFO: Loading module hook "hook-pkg_resources.py"...
8348 INFO: Processing pre-safe import module hook   win32com
8629 INFO: Looking for ctypes DLLs
8669 WARNING: library Advapi32 required via ctypes not found
8670 WARNING: library setupapi required via ctypes not found
8725 WARNING: library setupapi.dll required via ctypes not found
8734 WARNING: library ole32 required via ctypes not found
8797 INFO: Analyzing run-time hooks ...
8804 INFO: Including run-time hook 'pyi_rth_pkgres.py'
8806 INFO: Including run-time hook 'pyi_rth_multiprocessing.py'
8822 INFO: Looking for dynamic libraries
8920 INFO: Looking for eggs
8920 INFO: Using Python library /Users/api_team/miniconda3/envs/nrfutil/lib/libpython2.7.dylib
8926 INFO: Warnings written to /Users/api_team/pc-nrfutil/build/nrfutil/warn-nrfutil.txt
8972 INFO: Graph cross-reference written to /Users/api_team/pc-nrfutil/build/nrfutil/xref-nrfutil.html
9151 INFO: Appending 'datas' from .spec
9155 INFO: checking PYZ
9155 INFO: Building PYZ because PYZ-00.toc is non existent
9156 INFO: Building PYZ (ZlibArchive) /Users/api_team/pc-nrfutil/build/nrfutil/PYZ-00.pyz
9728 INFO: Building PYZ (ZlibArchive) /Users/api_team/pc-nrfutil/build/nrfutil/PYZ-00.pyz completed successfully.
9889 INFO: checking PKG
9889 INFO: Building PKG because PKG-00.toc is non existent
9889 INFO: Building PKG (CArchive) PKG-00.pkg
16526 INFO: Building PKG (CArchive) PKG-00.pkg completed successfully.
16577 INFO: Bootloader /Users/api_team/miniconda3/envs/nrfutil/lib/python2.7/site-packages/PyInstaller/bootloader/Darwin-64bit/run
16577 INFO: checking EXE
16578 INFO: Building EXE because EXE-00.toc is non existent
16578 INFO: Building EXE from EXE-00.toc
16579 INFO: Appending archive to EXE /Users/api_team/pc-nrfutil/dist/nrfutil
16594 INFO: Fixing EXE for code signing /Users/api_team/pc-nrfutil/dist/nrfutil
16599 INFO: Building EXE from EXE-00.toc completed successfully.
```

bihanssen, 2019-03-07:
@blasagna, I'm not able to reproduce the issue on this setup: Python 2.7.10, PyInstaller 3.4, macOS 10.14.3 (Platform: Darwin-18.2.0-x86_64-i386-64bit).
Haven't tried on other setups so far. Not sure if either Python version or macOS version could make a difference.

blasagna, 2019-03-08:
Thanks @bihanssen ! I tried with Python 2.7.10 and everything works now. Would it be possible to add a note on known working versions of Python and pyinstaller to the documentation?

PR #214 (2019-03-13): Update README.md
Add comment about PyInstaller and Python version. Closes #212.
Remove dead link to pc-ble-driver doc.
closed 2019-03-13 by bihanssen