NordicSemiconductor/pc-nrfutil #219: VirusTotal warnings!

E3V3A, 2019-03-26:
Uploading the latest release 5.1.0 for Windows, gives a number of hits with VT. 
I'm not sure what package mechanism you're using or if your dev machine is infected, but that is never a good sign. 

[image]

E3V3A, 2019-03-26:
May alse be related to unsolved (?) issue in different repo:
* https://github.com/NordicSemiconductor/pc-nrfconnect-core/issues/253

bihanssen, 2019-10-18:
We have tested the application and build machines with our AV tools and haven't found any indications on there being any real findings. We therefore consider the VirusTotal hits to be false positives. Closing the issue.

closed 2019-10-18 by bihanssen