NordicSemiconductor/pc-nrfutil #220: Please update to Python3

E3V3A, 2019-03-27:
Please consider updating your code to use Python3. Python2 is increasingly incompatible with many new updates of external packages.

bihanssen, 2019-03-28:
Hi, we plan to update to support python 3 when the underlying dependency pc-ble-driver-py gets python 3 support.

E3V3A, 2019-03-28:
Cool. Where is that driver hosted? Perhaps I can help push it forward?

bihanssen, 2019-03-29:
@E3V3A the updates are located in the https://github.com/NordicSemiconductor/pc-ble-driver-py/tree/feature/s132v5_py3 branch.
It is mostly finished. The remaining work is mostly to get compilation and CI running with all combinations of supported Python versions and operating systems.

goldwake, 2019-05-27:
It looks like [python 3.7 support has now been added to pc-ble-driver-py](https://github.com/NordicSemiconductor/pc-ble-driver-py/commit/94c7f440057e279ca4772d92133356a2b4d6c858)

Langhalsdino, 2019-07-08:
The python3 feature branch is a good start to make the lib work with python3.
We got the serial bootloader flashing to work with python3. 

We needed to make a few tweaks to the code, to do so. 

Checkout:
https://github.com/NordicSemiconductor/pc-nrfutil/pull/234
https://github.com/NordicSemiconductor/pc-nrfutil/tree/feature/convert-to-py3

dmascord, 2019-08-26:
In addition to the branch convert-to-py3, the following small changes are needed while attempting to build a bootloader and flashing it.
[
[python3.txt](https://github.com/NordicSemiconductor/pc-nrfutil/files/3539320/python3.txt)

](url)

E3V3A, 2019-09-04:
@Langhalsdino 
@dmascord

Why don't you post PR for these changes?

Langhalsdino, 2019-09-04:
@E3V3A We did ;) 
https://github.com/NordicSemiconductor/pc-nrfutil/pull/234

E3V3A, 2019-09-04:
So lets pray for some merges...

bihanssen, 2019-10-18:
nrfutil has been released as alpha with python 3 support: https://pypi.org/project/nrfutil/#history
Non-alpha version to be released soon.
Closing this issue.

closed 2019-10-18 by bihanssen