NordicSemiconductor/pc-nrfutil #222: Constants issue in 5.1.0

MattiasErikssonAxentia, 2019-04-01:
Hi and thanks for a great tool.
When using latest release 5.1.0 I get an error: ModuleNotFoundError: No module named 'constants'
After i manually install 'constants' with "pip install constants" I get the error: ImportError: cannot import name 'DIGCF_PRESENT' from 'constants' (c:\python37\lib\site-packages\constants.py)
on line nordicsemi\lister\windows\lister_win32.py", line 43, in <module> from constants import DIGCF_PRESENT, DEVPKEY, DIGCF_DEVICEINTERFACE

I have tried pre-compiled .exe file and installing via pip (I'm running on Windows 10 with python 3.7.1).

If I instead install version 5.0.0 I works good: pip install nrfutil==5.0.0
Would be good if you can check it out.
Best regards
Mattias Eriksson

bihanssen, 2019-04-01:
Hi,
> I'm running on Windows 10 with python 3.7.1
nrfutil does not currently support Python 3 I'm afraid. Use Python 2.7 instead. Py 3 support is on the roadmap.

MattiasErikssonAxentia, 2019-04-01:
Sorry, I realized this also afterwards.. Since all your tutorials simply says "pip install nrfutil" I just thought that there would not be any particular requirement that pip would not handle.
But I realize now that this is no the case. Now I just have to figure out how to handle two versions of python and pip in parallel.

closed 2019-04-01 by MattiasErikssonAxentia

bihanssen, 2019-04-01:
I agree, there should be some version checking in the pip install procedure. We'll look into adding some kind of check. Thanks for the input.

aguaviva, 2020-11-06:
I am getting this error, still no support for Python 3? :)

kashishkhullar, 2021-06-30:
> 2021... Seems that this error still present. Saw that on python 3.9

I got that too. Then I downgraded to 5.0.0.