NordicSemiconductor/pc-nrfutil #228: DFU trigger interface broke serial transport over arbitrary com port

bearsh, 2019-06-11:
In our product, for now, we tunnel the serial line from the pc to the nrf52 over our main microcontroller to implement dfu. This works so far. But since the integration of the *DFU Trigger* this no longer works.

To detect if a bootloader is active, the `DfuTransportSerial` class calls `__is_device_in_bootloader_mode()` which hard codes some usb vid/pid for known usb-to-uart devices (https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/dfu_transport_serial.py#L347) but makes it impossible to use a arbitrary usb-to-uart bridge.

One fix for this I can think of is to be able to tell the `DfuTransportSerial` object that the bootloader mode **is** active, hence to overwrite/disable the internal test.

mmbedded, 2020-01-08:
As far I understand  `__ensure_bootloader()` inside open method of DfuTransportSerial, should not be called in serial mode, but only in usb-serial mode. Is that correct? 
In serial mode `do_ping `is used for checking if target is in bootloader.
So why not call a function that function under statement `if not self.do_ping`

kenr, 2023-01-17:
We do not have capacity to maintain and release new versions of this utility.
A new utility with the same name will cover this feature. It can be found here: https://www.nordicsemi.com/Products/Development-tools/nrf-util

closed 2023-01-17 by kenr