NordicSemiconductor/pc-nrfutil #241: Can not send image over Zigbee with precompiled nrfutil

sebastiandraus, 2019-09-10:
It is impossible to send image over Zigbee when precompiled nrfutil is used.
Nrfutil used: release 5.2. downloaded form https://github.com/NordicSemiconductor/pc-nrfutil/releases
The error returned by nrfutil:

> .\nrfutil.exe dfu zigbee -f 6996fe001dd04620bb772a07e9df8c46_temporary_hex.zigbee -snr 683728386 --channel 12
> Traceback (most recent call last):
>   File "nordicsemi\__main__.py", line 1464, in <module>
>   File "click\core.py", line 764, in __call__
>   File "click\core.py", line 717, in main
>   File "click\core.py", line 1137, in invoke
>   File "click\core.py", line 1137, in invoke
>   File "click\core.py", line 956, in invoke
>   File "click\core.py", line 555, in invoke
>   File "nordicsemi\__main__.py", line 1418, in zigbee
>   File "nordicsemi\zigbee\ota_flasher.py", line 104, in fw_check
>   File "nordicsemi\zigbee\ota_flasher.py", line 94, in _fw_check
>   File "nordicsemi\zigbee\ota_flasher.py", line 89, in verify
>   File "site-packages\wrapt\decorators.py", line 440, in _synchronized
>   File "site-packages\pc_ble_driver_py\ble_driver.py", line 1115, in call_cmd
>   File "subprocess.py", line 219, in check_output
> subprocess.CalledProcessError: Command '['nrfjprog', '--snr', '683728386', '--verify', 'C:\\Users\\sedr\\AppData\\Local\\Temp\\_MEI25~1\\nordicsemi\\zigbee\\hex\\ota.hex', '--family', 'NRF52']' returned non-zero exit status 51
> [4220] Failed to execute script \_\_main\_\_

bihanssen, 2020-01-08:
For the record, I can confirm that the issue is reproducible, also with latest precompiled version of nrfutil v6.0.1. The issue seems to occur in a call to nrfjprog.

cmarignier, 2020-04-07:
Same output for me with the 6.0.1 version for Windows. Using an nRF52840-Preview-DK or nRF52840-DK shows the same output.

bihanssen, 2020-04-24:
Fixed in PR 320, closing.

closed 2020-04-24 by bihanssen