NordicSemiconductor/pc-nrfutil #246: Python3 errors: None vs int, raw bytes

chrysn, 2019-09-13:
The feature/convert-to-py3 branch (as well as apic-ai's version which I actually use) runs into several errors when invoked with `pkg generate --hw-version 52 --sd-req 0x00 --aplication-version 1 --application some.zip` due to missing Python 3 conversion steps at the following locations:

* `nordicsemi/__main__.py` lines 883 and 887: `zigbee_ota_{min,max}_hw_version` is compared to an integer but can be None, you may want to add a check for whether it `is not None and` the old condition holds.

* `nordicsemi/dfu/package.py` line 442 appends `''` to boot_validation_bytes_array which is later expected to hold bytes. You may want to append `b''` here.

(To make this findable when people are faced with the error messages: they were `TypeError: '>' not supported between instances of 'NoneType' and 'int'` and `TypeError: '' has type str, but expected one of: bytes`, respectively).

I can't make this into a pull request because of the CLA in combination with the non-OSI license, but I'm confident that you won't have much work turning the above into an applicable patch.

bihanssen, 2019-09-13:
Thanks for sharing. We'll incorporate your findings.

alwa-nordic, 2019-10-15:
Closing this issue as already fixed. This slightly corrected invocation works on current master:
```
python -m nordicsemi pkg generate --hw-version 52 --sd-req 0x00 --application-version 1 --application ./nordicsemi/dfu/tests/firmwares/foo.hex some.zip
python -m nordicsemi pkg display some.zip
```

closed 2019-10-15 by alwa-nordic